ros-perception/vision_opencv #60: cv_bridge for rosjava

talregev, 2015-03-21:
can it made cv_bridge for [rosjava](https://github.com/rosjava)?

vrabaud, 2015-03-22:
@damonkohler, OpenCV has support for Java these days. Do you have anything planned for a Java cv_bridge ?

talregev, 2015-07-25:
I have an initial code, but it not efficient. and for converting raw images, i get memory out  exceptions.

vrabaud, 2015-08-02:
are you using the OpenCV java bridge ? Should that code belong to the cv_bridge java ? @stonier, any feedback on where that should belong ?

talregev, 2015-08-02:
@vrabaud there is not such a thing OpenCV java bridge. but i want to help to create it.
I compile opencv in android project and convert images of opencv to ros msg.
So i have code that working. 
I publish here, and rosjava hopefully some one will read this and the implantation of it. When i started there is no code at all, and it very difficult to do it by your own.

vrabaud, 2015-08-03:
For OpenCV and Java, I was referring to: http://docs.opencv.org/doc/tutorials/introduction/desktop_java/java_dev_intro.html and http://packages.ubuntu.com/trusty/libopencv2.4-java
Not sure if that can help.

talregev, 2015-08-03:
@vrabaud it ok there will be java code that compile on android studio, make a jar file that other can use there directory for convert opencv images for ros msg like cv_bridg c++ and python dose?

vrabaud, 2015-08-04:
@stonier, I believe you can give us some insight on where we should put such a java OpenCV bridge. Thx

vrabaud, 2015-08-04:
@talregev , I am not sure I understand: do you want to create some java code in some fodler (in cv_bridge) that when compiled with Android studio gives a .jar file that you can use to convert messages to cv::Mat ?

talregev, 2015-08-04:
yes

talregev, 2015-08-05:
I publish the code [here](https://github.com/talregev/cv_bridge). 
It not complete, but most of the functionality is implemented.
I tried to do it as much as close to cv_bridge of cpp file.
It generate a .aar (Android library) file and not .jar, because opencv is for android and not java alone,
so i could only compile it to Android library.
Can it added to here and i can maintain it?

talregev, 2015-08-06:
@vrabaud

stonier, 2015-08-20:
@vrabaud I talked to him a couple of days ago on chat...

* he wants to use opencv in android
* and cv_bridge like functionality to bridge it to ros

His first thought I think was to keep everything as compatible with what is currently there, to the point of having it in the same repository (python and c++ co-exist in cv_bridge after all). The fact that there is code in sensor_msgs too was also confusing him.

End of the day, I pointed him to [rosjava_extras](https://github.com/rosjava/rosjava_extras) and [android_extras](https://github.com/rosjava/android_extras) as the best places to drop in some gradle subprojects to achieve what he wants. Mixed language packages with java are still difficult and every other non-msg java package exists separately anyway, so that follows the principle of least surprise.

talregev, 2015-09-03:
@vrabaud
You can close this issue. 
Already implement cv_bridge for java.
I put it here:
https://github.com/talregev/android_cv_bridge.

closed 2015-09-04 by vrabaud