ros-perception/vision_opencv #61: gtest dependency

lepalom, 2015-05-26:
This packages needs libgtest-dev to be built. Please add it to package.xml

vrabaud, 2015-05-26:
why package exactly ? It should be handled by catkin or the ENABLE_TESTING variable. Which error do you get.

lepalom, 2015-05-26:
cv_bridge

it's a hidden cmake error that makes the cv_bridge package failed to compile. It's solved installed this package. I'm working with Debian Jessie, compiling ros indigo, full desktop. I'm  using: 
catkin_make_isolated

The error:
==> Processing catkin package: 'cv_bridge'
==> Building with env: '/srv/robotica/ros/devel_isolated/control_msgs/env.sh'
==> cmake /srv/robotica/ros/src/vision_opencv/cv_bridge -DCATKIN_DEVEL_PREFIX=/srv/robotica/ros/devel_isolated/cv_bridge -DCMAKE_INSTALL_PREFIX=/srv/robotica/ros/install_isolated -G Unix Makefiles in '/srv/robotica/ros/build_isolated/cv_bridge'
-- Using CATKIN_DEVEL_PREFIX: /srv/robotica/ros/devel_isolated/cv_bridge
-- Using CMAKE_PREFIX_PATH: /srv/robotica/ros/devel_isolated/control_msgs;/srv/robotica/ros/install_isolated
-- This workspace overlays: /srv/robotica/ros/devel_isolated/control_msgs;/srv/robotica/ros/install_isolated
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /srv/robotica/ros/build_isolated/cv_bridge/test_results
CMake Error at /usr/share/catkin/cmake/test/gtest.cmake:120 (add_subdirectory):
  add_subdirectory given source "/usr/src/gtest" which is not an existing
  directory.
Call Stack (most recent call first):
  /usr/share/catkin/cmake/all.cmake:147 (include)
  /usr/share/catkin/cmake/catkinConfig.cmake:20 (include)
  CMakeLists.txt:4 (find_package)

CMake Error at /usr/share/catkin/cmake/test/gtest.cmake:122 (set_target_properties):
  set_target_properties Can not find target to add properties to: gtest
Call Stack (most recent call first):
  /usr/share/catkin/cmake/all.cmake:147 (include)
  /usr/share/catkin/cmake/catkinConfig.cmake:20 (include)
  CMakeLists.txt:4 (find_package)

-- Found gtest sources under '/usr/src/gtest': gtests will be built
CMake Warning at /usr/share/catkin/cmake/test/nosetests.cmake:97 (message):
  nosetests not found, Python tests can not be run (try installing package
  'python-nose')
Call Stack (most recent call first):
  /usr/share/catkin/cmake/all.cmake:147 (include)
  /usr/share/catkin/cmake/catkinConfig.cmake:20 (include)
  CMakeLists.txt:4 (find_package)

-- catkin 0.6.14
-- Boost version: 1.55.0
-- Found the following Boost libraries:
--   python
-- skipping nosetests(enumerants.py) in project 'cv_bridge'
-- Configuring incomplete, errors occurred!
See also "/srv/robotica/ros/build_isolated/cv_bridge/CMakeFiles/CMakeOutput.log".
See also "/srv/robotica/ros/build_isolated/cv_bridge/CMakeFiles/CMakeError.log".
<== Failed to process package 'cv_bridge': 
  Command '['/srv/robotica/ros/devel_isolated/control_msgs/env.sh', 'cmake', '/srv/robotica/ros/src/vision_opencv/cv_bridge', '-DCATKIN_DEVEL_PREFIX=/srv/robotica/ros/devel_isolated/cv_bridge', '-DCMAKE_INSTALL_PREFIX=/srv/robotica/ros/install_isolated', '-G', 'Unix Makefiles']' returned non-zero exit status 1

Reproduce this error by running:
==> cd /srv/robotica/ros/build_isolated/cv_bridge && /srv/robotica/ros/devel_isolated/control_msgs/env.sh cmake /srv/robotica/ros/src/vision_opencv/cv_bridge -DCATKIN_DEVEL_PREFIX=/srv/robotica/ros/devel_isolated/cv_bridge -DCMAKE_INSTALL_PREFIX=/srv/robotica/ros/install_isolated -G 'Unix Makefiles'

Command failed, exiting.

vrabaud, 2015-05-28:
Do you have a folder /usr/src/gtest ? What's inside ? can you locat gtest.cc on your machine ? Thx.

lepalom, 2015-05-28:
1) Yes, I have a folder /usr/src/gtest
2) What's inside?
ls -la /usr/src/gtest
total 28
drwxr-xr-x 4 root root 4096 May 27 13:51 .
drwxr-xr-x 3 root root 4096 May 27 13:51 ..
-rw-r--r-- 1 root root 9120 Apr 23  2014 CMakeLists.txt
drwxr-xr-x 2 root root 4096 May 27 13:51 cmake
drwxr-xr-x 2 root root 4096 May 27 13:51 src
3) gtest.cc is in /usr/src/gtest/src
It's a default debian installation (sid/stretch), in a clean chroot.

The bug is that you only haveto take care that libgtest-dev is installed. That's all, if not, cmake fails.

vrabaud, 2015-05-28:
I don't understand: catkin depends on gtest, so it should be installed (cf https://github.com/ros/catkin/blob/indigo-devel/package.xml#L26). Now, you have a custom catkin in  /usr/share/catkin/ . I am not sure on how you installed it but you should also take care of the runtime dependencies of the packages you install by hand (probably using rosdep), otherwise, anything crazy could happen like here.

lepalom, 2015-05-28:
> I don't understand: catkin depends on gtest, so it should be installed. 

Well, they put gtest, but it should be libgtest-dev. Maybe it's a catkin bug.

> Now, you have a custom catkin in /usr/share/catkin/ . I am not sure on how you installed it but you 
> should also take care of the runtime dependencies of the packages you install by hand.

No, a debian package especific for debian. I'll check  the gtest dependency you have commented.

jspricke, 2015-05-29:
gtest is an rosdep alias for libgtest-dev (cf. https://github.com/ros/rosdistro/blob/master/rosdep/base.yaml#L845), so I guess thins can be closed.

vrabaud, 2015-06-03:
closing, I hope the explanations given were clear enough. Thx @jspricke

closed 2015-06-03 by vrabaud