ros-perception/vision_opencv #66: toCvCopy Problems in Jade

danthony06, 2015-07-26:
I am following the C++ cv_bridge tutorial found here:

http://wiki.ros.org/cv_bridge/Tutorials/UsingCvBridgeToConvertBetweenROSImagesAndOpenCVImages

The provided code works fine in Ubuntu 14.04 with ROS Indigo. I am having problems with the same code in Ubuntu 15.04 with ROS Jade. If I attempt to run the tutorial node, I receive the following error when calling cv::namedWindow.

OpenCV Error: Unspecified error (The function is not implemented. Rebuild the library with Windows, GTK+ 2.x or Carbon support. If you are on Ubuntu or Debian, install libgtk2.0-dev and pkg-config, then re-run cmake or configure script) in cvNamedWindow, file /tmp/buildd/ros-jade-opencv3-3.0.0-1vivid-20150620-1427/modules/highgui/src/window.cpp, line 517
terminate called after throwing an instance of 'cv::Exception'
  what():  /tmp/buildd/ros-jade-opencv3-3.0.0-1vivid-20150620-1427/modules/highgui/src/window.cpp:517: error: (-2) The function is not implemented. Rebuild the library with Windows, GTK+ 2.x or Carbon support. If you are on Ubuntu or Debian, install libgtk2.0-dev and pkg-config, then re-run cmake or configure script in function cvNamedWindow

Using the instructions I found here, rebuilt OpenCV with gtk support:
http://stackoverflow.com/questions/30450602/opencv-3-0-python-imshow-error

Now when I run the tutorial node, the call to namedWindow() succeeds, I get an error when calling toCvCopy, that I am using an unknown/unsupported array type.

vrabaud, 2015-07-27:
you are using OpenCV3 : cv_bridge is compiled with OpenCV2 so you cannot have OpenCV3 on your path when cv_bridge. If you are using the cv_bridge package, you cannot have the opencv3 package.

danthony06, 2015-07-27:
Thanks for your help. Some other software in my system had installed OpenCV3, and that was confusing ROS.

closed 2015-07-27 by danthony06