ros-perception/vision_opencv #69: //include/opencv not found

presocratics, 2015-08-13:
I came across this issue when doing a full build of ROS from source in Fedora 21. I will first describe the error and then how I resolved it and then some details about my system.

The error:

    at /usr/local/ros/jade/share/image_proc/cmake/image_procConfig.cmake:106 (message):
    Project 'image_proc' specifies '//include/opencv' as an include dir, which
    is not found.  It does neither exist as an absolute directory nor in
    '/usr/local/ros/jade///include/opencv'.  Ask the maintainer 'Vincent Rabaud
    <vincent.rabaud@gmail.com>' to fix it.
    Call Stack (most recent call first):
    /usr/local/ros/jade/share/catkin/cmake/catkinConfig.cmake:75 (find_package)
    CMakeLists.txt:4 (find_package)

Somehow my opencv location "/usr/include/opencv" is being stored as "//include/opencv"

I'm not much of a cmake wizard, but what I decided to do is comment out "DEPENDS
OpenCV" in the catkin_package macro in the files listed below.

    src/image_pipeline/image_proc/CMakeLists.txt:21:  #DEPENDS OpenCV
    src/image_transport_plugins/compressed_image_transport/CMakeLists.txt:15:  #DEPENDS OpenCV
    src/image_transport_plugins/compressed_depth_image_transport/CMakeLists.txt:15:  #DEPENDS OpenCV
    src/vision_opencv/image_geometry/CMakeLists.txt:11:               #DEPENDS OpenCV
    src/vision_opencv/cv_bridge/CMakeLists.txt:18:  #DEPENDS OpenCV

With the above changes, I can successfully build jade on my machine. I'm new to
ROS, but fairly handy with Linux, so I can try to provide additional info if
needed.

Some details about my system:
Fedora 21 amd64
OpenCV installed via yum (/usr/include/opencv)
ROS installed to /usr/local/ros/jade/

Thanks for your work.

vrabaud, 2015-09-06:
sorry, I was out of town. @cottsay, have you ever encountered that on Fedora ? Thx

cottsay, 2015-09-06:
Thanks for tagging me in, @vrabaud.

I have indeed seen this before. It is a long-standing issue introduced by the `/usr` move in Fedora many releases ago. It is probably specific to Fedora, and is not a bug in `vision_opencv`

To summarize, since `/usr/lib64` and `/lib64` are linked, the routine that searches for the OpenCV libraries ends up finding them in `/lib64` before `/usr/lib64`. It then deduces that the corresponding include directory is at `../include`, thus yielding `//include`.

I thought that this issue was fixed in OpenCV (I think we're still seeing it in VTK, though). Make sure your packages are up-to-date.

IIRC, the workaround I used was to set `CMAKE_PREFIX_PATH` to `/usr` or something...I really don't remember exactly. You might try something like:
1. rm -rf build_isolated/vision_opencv
2. catkin_make_isolated --install --pkg vision_opencv -DCMAKE_PREFIX_PATH=/usr
3. catkin_make_isolated --install --from-pkg vision_opencv

...but YMMV. Be sure to re-add any other flags you used on `catkin_make_isolated`, but it probably wouldn't be a good idea to build any other packages with `-DCMAKE_PREFIX_PATH` set.

Thanks,

--scott

Related bugs:
- https://bugzilla.redhat.com/show_bug.cgi?id=917407
- https://bugzilla.redhat.com/show_bug.cgi?id=921271
- https://bugzilla.redhat.com/show_bug.cgi?id=1070428

vrabaud, 2015-09-06:
Wow, thx @cottsay , that is very precise.
@presocratics, did you compile your own OpenCV maybe ? Can you please confirm its version ? Thx.

presocratics, 2015-09-08:
@vrabaud I'm just running the yum provided opencv and opencv-devel packages. It's possible that opencv updated since this issue was posted but my current version is:
2.4.9-3

We are about to put ROS on a similarly setup Fedora laptop, so I can probably take a look at some of the workarounds suggested by @cottsay.

cottsay, 2015-09-08:
Looks like that is the latest version available in Fedora 21 [1].

I should mention that this issue manifests only when using Catkin to invoke the build. When we build the RPMs, we invoke `cmake` and `make` manually and don't see this (which is why it hasn't been much of a priority of mine to get it fixed).

--scott

[1] https://apps.fedoraproject.org/packages/opencv

danthony06, 2015-11-04:
FWIW, I'm seeing this issue with Fedora 23.

vrabaud, 2015-11-16:
is that issue on the catkin side then ?

danthony06, 2015-11-16:
I didn't poke into the root cause too much, or what the proper way of fixing this is, but this solution did resolve the issue for me. If there's something I can do to help diagnose the issue, let me know and I can try it.

http://answers.ros.org/question/199279/installation-from-source-fails-because-of-cv_bridge-include-dir/

cottsay, 2015-11-17:
This is a Fedora issue. I can repro it with a simple CMakeLists.txt file. It is NOT a `catkin` or `vision_opencv` problem.

ijnek, 2022-09-12:
I recently became the maintainer of this repo, and I'm sorting through issues, closing stale ones. As this issue has not been active for a while, I'm going to close it due to inactivity. Please feel free to comment if it is still relevant and you would like to get it re-opened.

closed 2022-09-12 by ijnek