ros-perception/vision_opencv #72: Grayscale images

locke87, 2015-09-15:
Hej there,

thank you for this nice little package, it currently helps me playing around with some opencv-algorithms on rosbag recordings.

However, support for grayscale images would be great. I had to change the code at some points so it didn't crash on cv::cvtColor. The code (or diff) I used is 

      /// Convert image to gray and blur it
      + if ( frame.channels() > 1 ) {
         cv::cvtColor( frame, src_gray, cv::COLOR_RGB2GRAY );
      + } else {
      +   src_gray = frame;
      + }
      cv::blur( src_gray, src_gray, cv::Size(3,3) );

I fixed it in findContours, generalContours and HoughCircles, but I guess this could be useful in other nodelets as well. What's your opinion about this?

vrabaud, 2015-09-15:
right, that is useful: please make a pull request to handle those cases, thx.

locke87, 2015-09-15:
done

vrabaud, 2015-09-16:
thx, fixed in #73

closed 2015-09-16 by vrabaud