ros-perception/vision_opencv #78: Missing dependency upon nodelet

marv, 2015-10-06:
Shouldn't the package.xml specify a dependency upon nodelet? It's required by the opencv_apps package

I created a rosinstall file using rosinstall_generator which include vision_opencv, but catkin_make fails with:

-- +++ processing catkin package: 'opencv_apps'
-- ==> add_subdirectory(vision_opencv/opencv_apps)
-- Using these message generators: gencpp;geneus;genlisp;genpy
CMake Error at /home/marv/ros_base_ws/devel/share/catkin/cmake/catkinConfig.cmake:75 (find_package):
  Could not find a package configuration file provided by "nodelet" with any
  of the following names:

    nodeletConfig.cmake
    nodelet-config.cmake

  Add the installation prefix of "nodelet" to CMAKE_PREFIX_PATH or set
  "nodelet_DIR" to a directory containing one of the above files.  If
  "nodelet" provides a separate development package or SDK, be sure it has
  been installed.
Call Stack (most recent call first):
  vision_opencv/opencv_apps/CMakeLists.txt:4 (find_package)

-- Configuring incomplete, errors occurred!

vrabaud, 2015-10-06:
It's been there from the very beginning: https://github.com/ros-perception/vision_opencv/blob/indigo/opencv_apps/package.xml#L18

I'd say you need to clean your build folder and recompile.

marv, 2015-10-07:
Oh, you're right, I was looking at the package.xml in the vision_opencv folder - sorry!

I see what the actual problem is now. The rosinstall file I generated with

   rosinstall_generator --verbose --rosdistro jade --deps --wet-only --upstream compressed_image_transport

included `vision_opencv`, because `compressed_image_transport` depends on `cv_bridge`.
The subsequent catkin_make call then failed because it also wants to build all other packages from `vision_opencv` which dependencies haven't been resolved...

Not sure how to handle this properly, basically I just wanted to satisfy the dependencies of a package from Github and thought rosinstall_generator would do that for me :-/

vrabaud, 2015-10-07:
well, you indeed cannot ask for upstream ```upstream``` and only have cv_bridge. This option gets the upstream repo and just checks it at a specific tag. What you could do is remove that option: it would in that case only get it from the gbp repo. In your case, you can also replace that entry in your config file with:
```
- git:
    local-name: vision_opencv/cv_bridge
    uri: https://github.com/ros-gbp/vision_opencv-release.git
    version: release/jade/cv_bridge/1.11.8-0
```

closed 2015-10-07 by vrabaud