ros-perception/vision_opencv #79: Bloom Release of vision_opencv for OpenCV3?

rethink-imcmahon, 2015-10-19:
I realize it is possible to compile `cv_bridge` and the `image_pipeline` for OpenCV3 [as mentioned in the ROS Answers post](http://answers.ros.org/question/213925/opencv30-ros-indigo/?answer=213928#post-id-213928), but is it possible to do a Debian release built against OpenCV3? This may require specifying a separate install `ros-indigo-cv3-bridge` package from `vision_opencv`, but there is a precedent for this sort of thing with `gazebo_ros_packages` making releases depending on different versions of Gazebo (2.2, 5.0, etc.). Any thoughts on this level of Debian cross OpenCV(2<->3) support?

chris-smith, 2015-10-19:
:+1: This would be nice to have. Indigo's got a lot of life left and it'd be nice to have better opencv3 support.

vrabaud, 2015-10-20:
I need to send an email to ROS users so that we brainstorm but that would be mostly problematic: having two versions of an executable is one thing, having two libraries at runtime would require a lot of complexity, especially at compile time (we would probably have to fumble with the RPATH if we want to make it proper). We could also have it be static only, or rename the OpenCV3 libraries to make them unique, that's probably the easiest for now.

vrabaud, 2015-11-16:
There is only one version of ```gazebo_ros_pkgs``` per ROS distro no ?
Ok, so here are the usual problems when having several version of a library for dev:
- you can only have one header for a given library so we would need to rename the headers of OpenCV3 to keep the OpenCV2 installed.
- ROS OpenCV3 packages are not following the proper way Debian is doing it: the .so files should be the same for both packages, they should just point to .so.2 or .so.3 when installing one over the other. ROS adds .so. The way I fix that right now is by renaming the libopencv_core.so to libopencv_core3.so (it's only been two weeks)

And yes, all that is fixable but takes resources. Keep an OpenCV3 version in Indigo would mean renaming things, hence yet another version to maintain: cv_bridge OpenCV2 for Indigo, cv_bridge OpenCV3 for Indigo with renaming, cv_bridge for the default OpenCV in Kinetic (most likely OpenCV3 but to be discussed at https://groups.google.com/forum/#!topic/ros-sig-perception/K5__71SX7eU ).

The use case is also not clear: yes, it's cool to have but there is no clear usage yet. When I see how hard it is for people to update their package to Jade, I am sure it will take them time to write code compatible with OpenCV3. And then ROS2 will come in ...

To summarize: yes, I could create such a package. No, I will not maintain it :) But the reason is valid and I could help doing it. Also, my opinion could change, if at some point, OpenCV3 has something that would be tremendously useful to ROS but I do not see anything yet that is a game changer.

ijnek, 2022-09-12:
I recently became the maintainer of this repo, and I'm sorting through issues, closing stale ones. As this issue has not been active for a while, I'm going to close it due to inactivity. Please feel free to comment if it is still relevant and you would like to get it re-opened.

closed 2022-09-12 by ijnek