ros-perception/vision_opencv #94: [cv_bridge] Colorized depth image with cvtColorForDisplay?

wkentaro, 2015-12-20:
like http://stackoverflow.com/questions/13840013/opencv-how-to-visualize-a-depth-image

[image]

How should I apply that option? Do you have any idea?

vrabaud, 2015-12-20:
Well, to get more complex options, we need to have a struct that contains display options (right now, we only have a few arguments that we pass along). So, just create a struct for display options and it should have members like: do_dynamic_scaling, min_image_value, max_image_value, color_map (which would be -1 by default but JET in your case). No ?

vrabaud, 2015-12-20:
Well, feel free to break the API :) I am sure we'll be able to handle more cases with that.

wkentaro, 2015-12-20:
All right :+1:

wkentaro, 2016-05-29:
done https://github.com/ros-perception/vision_opencv/pull/130

closed 2016-05-29 by wkentaro