ros-perception/vision_opencv #96: conflict with cv3.0?

zhudelong, 2015-12-22:
I did step by step but got such error. 
OpenCV Error: The function/feature is not implemented (Unknown/unsupported array type) in create, file /home/zhudelong/Downloads/Caffe/opencv-3.0.0/modules/core/src/matrix.cpp, line 2498
terminate called after throwing an instance of 'cv::Exception'
  what():  /home/zhudelong/Downloads/Caffe/opencv-3.0.0/modules/core/src/matrix.cpp:2498: error: (-213) Unknown/unsupported array type in function create

Aborted (core dumped)

and I am very sure it is this sentence that goes wrong
cv_ptr = cv_bridge::toCvCopy(msg, sensor_msgs::image_encodings::BGR8);
Who have any idea? Thank you very much

k-okada, 2015-12-23:
OpenCV3 is also tested in Travis, so please check any difference between
your compile log and the one from the Travis ->
https://travis-ci.org/ros-perception/vision_opencv/branches

vrabaud, 2016-01-02:
@zhudelong , you are using a custom OpenCV3. I'll close this specific bug but feel free to open up a new one if you can provide a piece of code that crashes with the official OpenCV 3.1 from ROS. Thx.

closed 2016-01-02 by vrabaud