ros-perception/vision_opencv #106: procps which is a test dependency is not available on OS X (or any of the BSD's)

spaghetti-, 2016-01-22:
Defined here: https://github.com/ros-perception/vision_opencv/blob/indigo/opencv_apps/package.xml#L34

Commited by @k-okada at https://github.com/ros-perception/vision_opencv/commit/99f5a4d61771426dd3a2db35473a2a9aa9105806

I couldn't find any uses for the program in the tests. What exactly is the reason for this change?

Thanks

k-okada, 2016-01-22:
> Commited by @k-okada <https://github.com/k-okada> at 99f5a4d

humm, I'm very sorry that I can't recall why I added this, #107 is the PR
to remove this.
I have no knowledge why procps is not available on BSD machine, but you can
add dummy entry to rosdep file, for example
```
procps:
  macports: []
```
to avoid this type of error

spaghetti-, 2016-01-22:
@k-okada no problem, just ran into this issue and I actually had a PR in progress with the same change, but looks like you've got it. Thanks for the quick change.

k-okada, 2016-01-22:
I see, I used this for `watch` program at https://github.com/ros-perception/vision_opencv/blob/indigo/opencv_apps/test/test-lk_flow.test#L17, periodically update optical flow. What is the equivalent program for OS X?

spaghetti-, 2016-01-22:
Homebrew seems to have https://gitlab.com/procps-ng/procps which is essentially the same. Maybe you can revert the merge and I'll submit a PR to rosdistro with this as required for OS X?

spaghetti-, 2016-01-22:
However, I do apologize for not doing my research first and going purely based off /proc.

trainman419, 2016-01-25:
This showed up on ROS Answers: http://answers.ros.org/question/224956/no-definition-of-procps-for-os-osx/

This does not appear to be fixed; #107 was closed without merging, and nothing was submitted to rosdep.

I think https://github.com/ros-perception/vision_opencv/blob/indigo/opencv_apps/test/test-lk_flow.test#L17 can be better done using `respawn="true"` and `respwan_delay=2` instead of relying on a linux-specific tool.

BluFoot, 2016-01-26:
I am still having this issue.

BluFoot, 2016-01-26:
I was able to fix it buy removing a line from vision_open_cv/opencv_apps/package.xml:
  <test_depend>procps</test_depend>

vrabaud, 2016-01-27:
All, what is the best solution ?

k-okada, 2016-01-27:
add http://braumeister.org/formula/watch to rosdistro?

k-okada, 2016-01-27:
I do not have OSX machine, so need someone who actually test if  https://github.com/ros/rosdistro/pull/10360 works

trainman419, 2016-01-27:
I think the better solution is to use the options that are already part of roslaunch; instead of introducing an additional dependency. I will submit a pull request.

trainman419, 2016-01-27:
Proposed fix which removes the procps dependency and uses roslaunch instead: #109

vrabaud, 2016-01-27:
fixed by #109 . Thx @trainman419  !

closed 2016-01-27 by vrabaud