ros-perception/vision_opencv #108: Question regarding OpenCV3, optflow.hpp and opencv-contrib dependency

bchretien, 2016-01-26:
Hi!

I'm maintaining ROS packages for Arch Linux, and a user (cc @Caopi) [reported](https://aur.archlinux.org/packages/ros-indigo-opencv-apps/) the following problem: when compiling `opencv_apps` with OpenCV3 installed,  the compilation failed for `simple_flow_nodelet.cpp` because `optflow.hpp` is missing (see the related commit: https://github.com/ros-perception/vision_opencv/commit/8ed5ff5c48b4c3d270cd8216175cf6a8441cb046). Currently, the only solution is to depend on the `opencv-contrib` package. Note that on Arch, the default OpenCV version is still OpenCV2.

The OpenCV2 version of `simple_flow_nodelet.cpp` relies on the standard `video/tracking.hpp` header, but the OpenCV3 version expects an `opencv-contrib` header. Are the `opencv-contrib` modules supposed to be standard for an OpenCV3 install? If not, that exe should be made optional with proper `opencv-contrib` checks in `CMakeLists.txt`, or the code should be modified to only depend on the standard OpenCV package.

vrabaud, 2016-01-26:
OpenCV3 is not officially supported in ROS.
Now, there is an OpenCV3 package that has been released that does not conflict with Ubuntu's OpenCV2. It has not been tested on Arch but that would be the one that should become the official one and yes, it includes opencv-contrib. How it is built is explained at the bottom of http://wiki.ros.org/action/show/vision_opencv?action=show&redirect=opencv
I am trying to get OpenCV3 as default in Kinetic and I need to create an OpenCV3 wiki page.
Any custom build of OpenCV2 or 3 is not supported.
Does that answer all your questions ?

Caopi, 2016-01-26:
Thanks for your answer vrabaud. I understand where you're coming from and appreciate your efforts to get OpenCV3 as default in Kinetic. 

The problem is not to get OpenCV 3 to work in Arch, there is a [package](https://aur.archlinux.org/packages/opencv-contrib/) to get it with all contrib libraries installed and it does work with ros-indigo. The rest is just a dependency issue or something that needs to be clarified in installation instructions.

But the main question I guess is why the OpenCV3 installation for ROS includes the contrib modules. Aren't those supposed to be unstable and to be merged later? It was my understanding that the standard build intended by the OpenCV developers was without contrib modules. Are they actually needed for the core ROS modules or do you just want to test them for later use already?

IIRC the whole ros-indigo-desktop-full was building fine for me with OpenCV3 without contrib modules before the mentioned commit, so I'm not sure if you need to sacrifice compatibility here for a sample file.

vrabaud, 2016-01-27:
contrib is a place where we set whatever we cannot maintain because we lack ressources (and it might have been part of the main OpenCV before). It's also where we put things that have an iffy license (sift) or that is not mature enough (GSOC usually).
Now, contrib was part of OpenCV 2.4. Still, Ubuntu did not include it. Whether it's one or the other repo  is not the issue.

One of the main motivations for getting OpenCV3 in ROS Kinetic is actually all the extra stuff from contrib: tracking, RGBD, SfM and many others are useful to robotics.

I will create a proper OpenCV3. Also, OpenCV is getting more modular CMake-wise in 3.2 or 3.3. That will help clarify what is needed for compilation.

Caopi, 2016-01-27:
Thanks for the clarification. I guess this can be closed then.

bchretien, 2016-01-27:
Thanks for the information, we'll follow your work on OpenCV3 and notify the Arch package maintainer once things stabilize.

@Caopi in the meantime, I can make a quick fix to add opencv-contrib as a dependency if `opencv >= 3` is detected. That should solve your issue, but as @vrabaud said, OpenCV3 is not officially supported yet, so other ROS packages may break down the line.

vrabaud, 2016-01-27:
Feel free to complete anything in here: http://wiki.ros.org/opencv3
(I hope added everything from this conversation)

closed 2016-01-27 by vrabaud

bchretien, 2016-01-27:
@Caopi I just uploaded `ros-indigo-opencv3` [to the AUR](https://aur.archlinux.org/packages/ros-indigo-opencv3). Feel free to give it a go and report any error.

k-okada, 2016-01-27:
ok,  I have created PR to check if OpenCV contains contrib package, see #110 and check if this work for you.

bchretien, 2016-01-27:
@Caopi you can test @k-okada's patch with [this PKGBUILD](https://gist.github.com/bchretien/ef37b2add0edb79b04e5).

Caopi, 2016-01-27:
It did compile on my machine with opencv3 from the opencv-contrib aur package, but as far as I can tell it did not detect it correctly. It does not try to build simple_flow_exe during my build, it only generates a header file.

Is there an easy way to output the contents of this variable during the build without adding another pull request to the repo?

But I also just realized that the simple_flow_exe is also not installed in the original version (see https://github.com/ros-perception/vision_opencv/blob/indigo/opencv_apps/CMakeLists.txt#L283-L284). Is this intended?

bchretien, 2016-01-27:
In CMake, you can print the value of a variable with:

```cmake
MESSAGE("SOME_VAR: ${SOME_VAR}")
```

For testing your changes, go to the build directory and use `cmake`/`make` rather than re-running `makepkg` from the root (that would squash any changes to the local files). Just make sure that the ROS environment variables are properly set.

Caopi, 2016-01-27:
This seems to be an issue with the AUR package. My lib components list looks like this:

-- OpenCV_LIB_COMPONENTS=opencv_xphoto;opencv_xobjdetect;opencv_ximgproc;opencv_xfeatures2d;opencv_tracking;opencv_text;opencv_surface_matching;opencv_structured_light;opencv_stereo;opencv_saliency;opencv_rgbd;opencv_reg;opencv_plot;opencv_optflow;opencv_line_descriptor;opencv_hdf;opencv_fuzzy;opencv_face;opencv_dpm;opencv_dnn;opencv_datasets;opencv_ccalib;opencv_bioinspired;opencv_bgsegm;opencv_aruco;opencv_viz;opencv_videostab;opencv_videoio;opencv_video;opencv_superres;opencv_stitching;opencv_shape;opencv_photo;opencv_objdetect;opencv_ml;opencv_imgproc;opencv_imgcodecs;opencv_highgui;opencv_flann;opencv_features2d;opencv_core;opencv_calib3d

It containts the seperate modules, but not "opencv_contrib". That's why it could not be recognized. If this is not how it's supposed to be done we might want to contact the package maintainer.

Otherwise, a check like this worked for me (i excluded the trailing semicolon because it might be the last package in the list) : 

if("${OpenCV_LIB_COMPONENTS}" MATCHES ";opencv_optflow")

But I think a check like this would not pass for OpenCV2. It would not break the installation, but the optflow file would not be included, even though in the optflow file there is a working case for opencv2.

You probably need to do another check around it to see if the user is even using OpenCV3 and only then check for contrib/optflow.

But since we agreed to accept the contrib as required and are trying to establish the ros opencv3 package with contrib installed, I'm not sure this is wise.