ros-perception/vision_opencv #115: [cv_bridge] .step is ignored in python

eric-wieser, 2016-02-27:
For example if I have padding bytes on the end of rows, this will fail

eric-wieser, 2016-02-27:
This is partially fixed by doing [this](https://github.com/eric-wieser/ros_numpy/blob/master/src/ros_numpy/image.py#L63-L67), but if you're aiming for cv compatibility, you might need an `np.copy` as well.

vrabaud, 2016-03-08:
why would the copy be needed ?

eric-wieser, 2016-03-08:
At one point, opencv required numpy arrays to have contiguous memory ordering. That might not be true in later releases. Step can lead to non-contiguous memory.

ijnek, 2022-09-12:
I recently became the maintainer of this repo, and I'm sorting through issues, closing stale ones. As this issue has not been active for a while, I'm going to close it due to inactivity. Please feel free to comment if it is still relevant and you would like to get it re-opened.

closed 2022-09-12 by ijnek