ros-perception/vision_opencv #120: Kinetic release

wjwwood, 2016-03-16:
@vrabaud what do we need to do to prepare this for a Kinetic release. I was just trying to build it from source and ran into a missing rosdep rule for `libopencv-dev`. Will that now be a ROS package dependency rather than using OpenCV from the system?

wjwwood, 2016-03-16:
For now I just manually installed `libopencv-dev`.

vrabaud, 2016-03-17:
it needs to be branched, simply, and have the rosdep key changed. Going away from the internet for 10 days on Sat so I'd better release it by then ....
That's the only one to branch, the other packages depending on OpenCV (e.g. image*) should depend on cv_bridge only so that they are not branched for a package.xml modification only.

wjwwood, 2016-03-17:
If you have time to do it is appreciate it. Otherwise leave me instructions and I'll do it.

vrabaud, 2016-03-17:
I'll try tonight but overall, all the core stuff compiles/passes tests and most likely works with OpenCV3 (I have not tested it all in real cases).
So vision_opencv needs to be branched, have the libopencv-dev key be replaced with opencv3 and that's it. Anything else should depend on cv_bridge as explained in http://wiki.ros.org/opencv3#package.xml as I don't want to do more branching.

wjwwood, 2016-03-17:
@vrabaud in that case we need an "opencv3" ROS package released into Kinetic right? That was the part I was wondering about (in terms of instructions for me).

vrabaud, 2016-03-17:
right, releasing it right now.

dirk-thomas, 2016-03-30:
The following needs to be addressed:
* the rosdep keys for the dependencies are still missing
* one package was ignored for the release and
* the repo has a circular dependency with another repo: https://github.com/ros-perception/vision_opencv/pull/103/files#r57967284

vrabaud, 2016-04-05:
@k-okada , I see no other solution but to move opencv_apps in its own repo. Are you ok with that ? Thx

k-okada, 2016-04-06:
Or how about removing imgae_proc/image_view from test_depends, we need them
to run the (current) test, but not need to compile opencv_apps itself.

vrabaud, 2016-05-07:
release made for Kinetic and opencv_apps moved to its own repo: https://github.com/ros-perception/vision_opencv/issues

closed 2016-05-07 by vrabaud