ros-perception/vision_opencv #122: [opencv_apps] I coundn't find structual analysis nodes when I used ros-indigo-opencv-apps:amd64/trusty 1.11.11-0trusty-20160321-200323-0700

kochigami, 2016-04-05:
Hi, 

When I tried ```rosrun opencv_apps convex_hull image:=/pepper_robot/naoqi_driver_node/camera/front/image_raw```, I got this error:  ```[rosrun] Couldn't find executable named convex_hull below /opt/ros/indigo/share/opencv_apps``` .

I use ubuntu 14.04 and ```ros-indigo-opencv-apps:amd64/trusty 1.11.11-0trusty-20160321-200323-0700```. 
I couldn't find ```find_contours```, ```convex_hull```, ```general_contours``` and ```contour_moments``` in ```/opt/ros/indigo/lib/opencv_apps``` in my environment. 
They are listed as stuructual analysis nodes in http://wiki.ros.org/opencv_apps .

The contents under ```/opt/ros/indigo/lib/opencv_apps``` are as follows: 
```
/opt/ros/indigo/lib/opencv_apps$ ls
camshift           hough_circles  segment_objects
edge_detection     hough_lines    simple_compressed_example
face_detection     lk_flow        simple_example
fback_flow         people_detect  watershed_segmentation
goodfeature_track  phase_corr
```

I tried ``` sudo apt-get update```, ```sudo apt-get upgrade```, ```sudo dpkg -r --force-depends ros-indigo-opencv-apps``` and ```sudo apt-get install ros-indigo-opencv-apps```, but the terminal said that the version of ```ros-indigo-opencv-apps``` is latest and the number of programs didn't change.

The content of my ```/etc/apt/sources.list.d/ros-latest.list``` is
```deb http://packages.ros.org/ros-shadow-fixed/ubuntu trusty main```.

Could anyone help me? 
I'm very sorry for my trouble.

k-okada, 2016-04-06:
Thanks you for report , I forget to install these binaries, fixed in #123.

kochigami, 2016-04-06:
Thank you very much for your help.

closed 2016-04-06 by kochigami