ros-perception/vision_opencv #138: Compiling error under inidgo branch, "‘UMatData’ does not name a type"

libing64, 2016-07-08:
I want to use the OpenCV3 with ros indigo, and some said that cv_bridge and image_pipeline should be recompiled. [http://answers.ros.org/question/213925/opencv30-ros-indigo/](http://answers.ros.org/question/213925/opencv30-ros-indigo/)

After installing OpenCV3 from source, I try to recompile vision_opencv, but I can not compile it successfully.
Can anyone provide some help or advise?  

```
/home/libing/catkin_ws/vision_opencv/cv_bridge/src/module_opencv3.cpp:97:5: error: ‘UMatData’ does not name a type
     UMatData* allocate(PyObject* o, int dims, const int* sizes, int type, size_t* step) const
     ^
/home/libing/catkin_ws/vision_opencv/cv_bridge/src/module_opencv3.cpp:110:5: error: ‘UMatData’ does not name a type
     UMatData* allocate(int dims0, const int* sizes, int type, void* data, size_t* step, int flags, UMatUsageFlags usageFlags) const
     ^
/home/libing/catkin_ws/vision_opencv/cv_bridge/src/module_opencv3.cpp:139:19: error: ‘UMatData’ has not been declared
     bool allocate(UMatData* u, int accessFlags, UMatUsageFlags usageFlags) const
                   ^
/home/libing/catkin_ws/vision_opencv/cv_bridge/src/module_opencv3.cpp:139:49: error: ‘UMatUsageFlags’ has not been declared
     bool allocate(UMatData* u, int accessFlags, UMatUsageFlags usageFlags) const
                                                 ^
/home/libing/catkin_ws/vision_opencv/cv_bridge/src/module_opencv3.cpp:144:21: error: ‘UMatData’ has not been declared
     void deallocate(UMatData* u) const
                     ^
/home/libing/catkin_ws/vision_opencv/cv_bridge/src/module_opencv3.cpp: In constructor ‘NumpyAllocator::NumpyAllocator()’:
/home/libing/catkin_ws/vision_opencv/cv_bridge/src/module_opencv3.cpp:94:39: error: ‘getStdAllocator’ is not a member of ‘cv::Mat’
     NumpyAllocator() { stdAllocator = Mat::getStdAllocator(); }
                                       ^
/home/libing/catkin_ws/vision_opencv/cv_bridge/src/module_opencv3.cpp: In member function ‘bool NumpyAllocator::allocate(int*, int, int) const’:
/home/libing/catkin_ws/vision_opencv/cv_bridge/src/module_opencv3.cpp:141:65: error: no matching function for call to ‘cv::MatAllocator::allocate(int*&, int&, int&) const’
         return stdAllocator->allocate(u, accessFlags, usageFlags);
                                                                 ^
/home/libing/catkin_ws/vision_opencv/cv_bridge/src/module_opencv3.cpp:141:65: note: candidate is:
In file included from /home/libing/catkin_ws/vision_opencv/cv_bridge/include/cv_bridge/cv_bridge.h:43:0,
                 from /home/libing/catkin_ws/vision_opencv/cv_bridge/src/module.hpp:22,
                 from /home/libing/catkin_ws/vision_opencv/cv_bridge/src/module_opencv3.cpp:3:
/usr/include/opencv2/core/core.hpp:1470:18: note: virtual void cv::MatAllocator::allocate(int, const int*, int, _Atomic_word*&, uchar*&, uchar*&, size_t*)
     virtual void allocate(int dims, const int* sizes, int type, _Atomic_word*& refcount,
                  ^
/usr/include/opencv2/core/core.hpp:1470:18: note:   candidate expects 7 arguments, 3 provided
/home/libing/catkin_ws/vision_opencv/cv_bridge/src/module_opencv3.cpp: In member function ‘void NumpyAllocator::deallocate(int*) const’:
/home/libing/catkin_ws/vision_opencv/cv_bridge/src/module_opencv3.cpp:149:41: error: request for member ‘userdata’ in ‘* u’, which is of non-class type ‘int’
             PyObject* o = (PyObject*)u->userdata;
                                         ^
/home/libing/catkin_ws/vision_opencv/cv_bridge/src/module_opencv3.cpp: At global scope:
/home/libing/catkin_ws/vision_opencv/cv_bridge/src/module_opencv3.cpp:158:16: error: cannot declare variable ‘g_numpyAllocator’ to be of abstract type ‘NumpyAllocator’
 NumpyAllocator g_numpyAllocator;
                ^
/home/libing/catkin_ws/vision_opencv/cv_bridge/src/module_opencv3.cpp:91:7: note:   because the following virtual functions are pure within ‘NumpyAllocator’:
 class NumpyAllocator : public MatAllocator
       ^
In file included from /home/libing/catkin_ws/vision_opencv/cv_bridge/include/cv_bridge/cv_bridge.h:43:0,
                 from /home/libing/catkin_ws/vision_opencv/cv_bridge/src/module.hpp:22,
                 from /home/libing/catkin_ws/vision_opencv/cv_bridge/src/module_opencv3.cpp:3:
/usr/include/opencv2/core/core.hpp:1470:18: note: 	virtual void cv::MatAllocator::allocate(int, const int*, int, _Atomic_word*&, uchar*&, uchar*&, size_t*)
     virtual void allocate(int dims, const int* sizes, int type, _Atomic_word*& refcount,
                  ^
/usr/include/opencv2/core/core.hpp:1472:18: note: 	virtual void cv::MatAllocator::deallocate(_Atomic_word*, uchar*, uchar*)
     virtual void deallocate(_Atomic_word* refcount, uchar* datastart, uchar* data) = 0;
                  ^
/home/libing/catkin_ws/vision_opencv/cv_bridge/src/module_opencv3.cpp: In function ‘bool pyopencv_to(PyObject*, cv::Mat&, ArgInfo)’:
/home/libing/catkin_ws/vision_opencv/cv_bridge/src/module_opencv3.cpp:329:7: error: ‘class cv::Mat’ has no member named ‘u’
     m.u = g_numpyAllocator.allocate(o, ndims, size, type, step);
       ^
/home/libing/catkin_ws/vision_opencv/cv_bridge/src/module_opencv3.cpp:329:63: error: no matching function for call to ‘NumpyAllocator::allocate(PyObject*&, int&, int [33], int&, size_t [33])’
     m.u = g_numpyAllocator.allocate(o, ndims, size, type, step);
                                                               ^
/home/libing/catkin_ws/vision_opencv/cv_bridge/src/module_opencv3.cpp:329:63: note: candidate is:
/home/libing/catkin_ws/vision_opencv/cv_bridge/src/module_opencv3.cpp:139:10: note: bool NumpyAllocator::allocate(int*, int, int) const
     bool allocate(UMatData* u, int accessFlags, UMatUsageFlags usageFlags) const
          ^
/home/libing/catkin_ws/vision_opencv/cv_bridge/src/module_opencv3.cpp:139:10: note:   candidate expects 3 arguments, 5 provided
/home/libing/catkin_ws/vision_opencv/cv_bridge/src/module_opencv3.cpp: In function ‘PyObject* pyopencv_from(const cv::Mat&)’:
/home/libing/catkin_ws/vision_opencv/cv_bridge/src/module_opencv3.cpp:352:12: error: ‘class cv::Mat’ has no member named ‘u’
     if(!p->u || p->allocator != &g_numpyAllocator)
            ^
/home/libing/catkin_ws/vision_opencv/cv_bridge/src/module_opencv3.cpp:358:33: error: ‘class cv::Mat’ has no member named ‘u’
     PyObject* o = (PyObject*)p->u->userdata;
                                 ^
make[2]: *** [cv_bridge/src/CMakeFiles/cv_bridge_boost.dir/module_opencv3.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [cv_bridge/src/CMakeFiles/cv_bridge_boost.dir/all] Error 2

```

vrabaud, 2016-07-08:
did you follow the instructions at http://wiki.ros.org/opencv3 ?

libing64, 2016-07-08:
@vrabaud Great thanks. 
After adding the "CMAKE_PREFIX_PATH" and "OpenCV_INCLUDE_DIRS" to the CMakeListsts.txt, it works now.

set(CMAKE_PREFIX_PATH "/home/libing/source/opencv3_install")
find_package(OpenCV 3 REQUIRED)
set(OpenCV_INCLUDE_DIRS "/home/libing/source/opencv3_install/include")

closed 2016-07-08 by libing64