ros-perception/vision_opencv #139: Migrate from standard apt libraries to this

B3J4y, 2016-07-20:
Hi,
I am not so familiar with ros and I want to migrate my project, which is nested in ros, from Opencv 2.4.x to Opencv 3.x . My assumption is that the standard apt ros-jade installation is based on Opencv 2.4.x because I could install the ros-jade-opencv3 package afterwards.

So If I am using opencv_bridge it will always be linked to opencv 2.4.x and this is obviousely crashing my process.

So, this is more a question than a bug: How can I cleanly replace the standard vision_opencv package with this one?

k-okada, 2016-07-21:
if your question is how to use opencv3 with vision_opencv, use kinetic(16.04), which uses opencv3 as an default or see 
https://github.com/ros-perception/vision_opencv/blob/indigo/.travis.yml#L6 to how to use opencv3 on indigo/jade (14.04)

vrabaud, 2016-07-21:
This is a question and is therefore more appropriate on http://answers.ros.org/
http://wiki.ros.org/opencv3 should have answers to all your questions. If not, comment in this thread. Thx.

closed 2016-07-21 by vrabaud