ros-perception/vision_opencv #148: Extra print to stdout in cv2_to_imgmsg with current ros-indigo-cv-bridge package for Ubuntu 14.04

francoisferland, 2016-08-09:
Hi,

There's an extra message being printed in the cv2_to_imgmsg around line 259 in core.py. The print statement seems to have been removed on a previous commit in the indigo branch (6f55a75), but it somehow made it in the package for Ubuntu 14.04 (at least for amd64). It can be seen in the source of the package from July 13:

http://packages.ros.org/ros/ubuntu/pool/main/r/ros-indigo-cv-bridge/

k-okada, 2016-08-09:
I think that's because of the source tree including the patch is not
released yet, we have to wait for next release

https://github.com/ros-perception/vision_opencv/commits/kinetic
https://github.com/ros-perception/vision_opencv/commits/indigo

vrabaud, 2016-08-09:
Thx for pointing that out. It has indeed been removed but not released as there is no emergency to release vision_opencv for now. If you think this is a strong bug (maybe it pollutes your terminal or logs), please let us know.

francoisferland, 2016-08-09:
I agree, it's not that strong of a bug since it doesn't really affect functionality and is pretty easy to fix locally. This is what I did on my installation. 

However, it does flood the terminal and logs pretty quickly, as it's quite common to use it at a camera's framerate.

patrickwang90, 2016-09-07:
Thanks for the info! Fixed the problem locally and definitely helps for debugging other issues. Much appreciated.

vrabaud, 2016-09-30:
Fixed in 1.12.2 and 1.11.14. Big sorry about that.

closed 2016-09-30 by vrabaud