ros-perception/vision_opencv #149: cv_bridge package missing imports

chrisspen, 2016-08-24:
In Indigo, this Python code worked fine:

     from cv_bridge import CvBridge, CvBridgeError

However, I recently did a fresh install of Kinetic and even though the install of cv_bridge reported no errors, now this import throws:

    ImportError: cannot import name CvBridge

Inspecting the cv_bridge module shows nothing:

```
>>> import cv_bridge
>>> dir(cv_bridge)
['__builtins__', '__doc__', '__file__', '__name__', '__package__', '__path__']

```

Has this package been changed in Kinetic, or have I mis-installed the package?

vrabaud, 2016-08-24:
You installed everything from packages ?

chrisspen, 2016-08-24:
I'm running on a Raspberry Pi 3, for which there isn't a stable Ubuntu image available, so I'm using Raspbian, for which there are no packages available, so I installed it from source per the [wiki's instructions.](http://wiki.ros.org/kinetic/Installation/Source)

However, I figured out the root problem. I'm using an overlay workspace to maintain some patched third-party packages, and when I called `inspect.getsourcefile(cv_bridge)`, I noticed it was returning a path to this overlay instead of Kinetic's default install location. Sure enough, somehow an older version of cv_bridge had been installed into the overlay, and that version apparently didn't populate __init__.py with the normal classes, yet I could still import them from the core subpackage.

Deleting that Python package from the overlay caused it to revert to the current version in Kinetic, and that fixed the error.

closed 2016-08-24 by chrisspen