ros-perception/vision_opencv #155: OpenCV 3 dependency - too big

vfdev-5, 2016-10-14:
Hello,
I wonder if it would be possible to create a smaller opencv3 dependency, instead of the present one. 
I needed to install `rgbd_launch` package, ros kinetic version, but when `apt-get` tells me that I need 425Mb of different packages, especially, `qt5webkit, qtdeclarative5, qt5quickwiget, vtk`  and even `libgdal-dev` and so on... Checking the dependencies, we see that `rgbd_launch` depends indirectly on `cv_bridge` and `image_transport` and both depends on `opencv3`.  No problem for dependencies on image format drivers, but `qt5` and `vtk` dependencies which draw themself another bunch of dependencies, makes it too big. 

Would it be better to have something like `ros-kinetic-opencv3-core` `ros-kinetic-opencv3-highgui` and other modules as for `opencv 2 ?

vrabaud, 2016-10-16:
I totally agree with the principle but it would take quite some effort because OpenCV is not meant to be released by piece: having several packages would require the appropriate FindFoo.cmake files to be created (e.g. Findopencv-gui.cmake). I do not have the time to create such files but I have pinged the OpenCV team so that this feature lands at some point.

StefanFabian, 2020-12-17:
Reviving this as the OpenCV 4 dependency is even bigger and it should be possible to use only the relevant opencv modules as exec depends.
I understand that it's kind of hard to reduce the build dependencies but I don't think it is necessary to also ship all of that with the debian package.
Currently it installs the following packages:
```
adwaita-icon-theme at-spi2-core autoconf automake autotools-dev cpp-8 dbus
  dbus-user-session dconf-gsettings-backend dconf-service dmsetup file
  fontconfig fontconfig-config fonts-dejavu-core gcc-8 gdal-data gfortran
  gfortran-8 gfortran-9 gir1.2-glib-2.0 glib-networking glib-networking-common
  glib-networking-services gsettings-desktop-schemas gstreamer1.0-plugins-base
  gtk-update-icon-cache hicolor-icon-theme humanity-icon-theme i965-va-driver
  ibverbs-providers intel-media-va-driver iso-codes javascript-common libaacs0
  libaec0 libaom0 libapparmor1 libargon2-1 libarmadillo9 libarpack2
  libatk-bridge2.0-0 libatk1.0-0 libatk1.0-data libatspi2.0-0 libavahi-client3
  libavahi-common-data libavahi-common3 libavcodec-dev libavcodec58
  libavformat-dev libavformat58 libavresample-dev libavresample4 libavutil-dev
  libavutil56 libbdplus0 libbluray2 libcaf-openmpi-3 libcairo-gobject2
  libcairo2 libcap2 libcap2-bin libcdparanoia0 libcfitsio8 libcharls2
  libchromaprint1 libcoarrays-dev libcoarrays-openmpi-dev libcodec2-0.9
  libcolord2 libcryptsetup12 libcups2 libdap25 libdapclient6v5 libdatrie1
  libdbus-1-3 libdc1394-22 libdc1394-22-dev libdconf1 libdevmapper1.02.1
  libdrm-amdgpu1 libdrm-common libdrm-intel1 libdrm-nouveau2 libdrm-radeon1
  libdrm2 libelf1 libepoxy0 libepsilon1 libevent-2.1-7 libevent-core-2.1-7
  libevent-dev libevent-extra-2.1-7 libevent-openssl-2.1-7
  libevent-pthreads-2.1-7 libexif-dev libexif-doc libexif12 libfabric1
  libfontconfig1 libfreetype6 libfreexl1 libfribidi0 libfyba0 libgcc-8-dev
  libgd3 libgdal26 libgdcm-dev libgdcm3.0 libgdk-pixbuf2.0-0
  libgdk-pixbuf2.0-bin libgdk-pixbuf2.0-common libgeos-3.8.0 libgeos-c1v5
  libgeotiff5 libgfortran-8-dev libgfortran-9-dev libgif7
  libgirepository-1.0-1 libgl1 libgl1-mesa-dri libgl2ps1.4 libglapi-mesa
  libglvnd0 libglx-mesa0 libglx0 libgme0 libgphoto2-6 libgphoto2-dev
  libgphoto2-l10n libgphoto2-port12 libgraphite2-3 libgsm1
  libgstreamer-plugins-base1.0-0 libgstreamer1.0-0 libgtk-3-0 libgtk-3-bin
  libgtk-3-common libharfbuzz0b libhdf4-0-alt libhdf5-103 libhdf5-openmpi-103
  libhwloc-dev libhwloc-plugins libhwloc15 libibverbs-dev libibverbs1 libice6
  libigdgmm11 libilmbase-dev libilmbase24 libip4tc2 libjbig-dev libjbig0
  libjpeg-dev libjpeg-turbo8 libjpeg-turbo8-dev libjpeg8 libjpeg8-dev
  libjs-jquery libjson-c4 libjson-glib-1.0-0 libjson-glib-1.0-common
  libkmlbase1 libkmldom1 libkmlengine1 libkmod2 liblcms2-2 liblept5 libllvm10
  libltdl-dev liblzma-dev libmagic-mgc libmagic1 libminizip1 libmp3lame0
  libmpg123-0 libnetcdf-c++4 libnetcdf15 libnl-3-200 libnl-3-dev
  libnl-route-3-200 libnl-route-3-dev libnspr4 libnss-systemd libnss3
  libnuma-dev libnuma1 libogdi4.1 libogg0 libopencv-calib3d-dev
  libopencv-calib3d4.2 libopencv-contrib-dev libopencv-contrib4.2
  libopencv-core-dev libopencv-core4.2 libopencv-dev libopencv-dnn-dev
  libopencv-dnn4.2 libopencv-features2d-dev libopencv-features2d4.2
  libopencv-flann-dev libopencv-flann4.2 libopencv-highgui-dev
  libopencv-highgui4.2 libopencv-imgcodecs-dev libopencv-imgcodecs4.2
  libopencv-imgproc-dev libopencv-imgproc4.2 libopencv-ml-dev libopencv-ml4.2
  libopencv-objdetect-dev libopencv-objdetect4.2 libopencv-photo-dev
  libopencv-photo4.2 libopencv-shape-dev libopencv-shape4.2
  libopencv-stitching-dev libopencv-stitching4.2 libopencv-superres-dev
  libopencv-superres4.2 libopencv-ts-dev libopencv-video-dev
  libopencv-video4.2 libopencv-videoio-dev libopencv-videoio4.2
  libopencv-videostab-dev libopencv-videostab4.2 libopencv-viz-dev
  libopencv-viz4.2 libopencv4.2-java libopencv4.2-jni libopenexr-dev
  libopenexr24 libopenjp2-7 libopenmpi-dev libopenmpi3 libopenmpt0 libopus0
  liborc-0.4-0 libpam-cap libpam-systemd libpango-1.0-0 libpangocairo-1.0-0
  libpangoft2-1.0-0 libpciaccess0 libpixman-1-0 libpmix2 libpng-dev
  libpng-tools libpng16-16 libpoppler97 libpq5 libproj15 libprotobuf17
  libproxy1v5 libpsm-infinipath1 libpsm2-2 libqhull7 libraw1394-11
  libraw1394-dev libraw1394-tools librdmacm1 librest-0.7-0 librsvg2-2
  librsvg2-common libsensors-config libsensors5 libshine3 libsigsegv2 libsm6
  libsnappy1v5 libsocket++1 libsoup-gnome2.4-1 libsoup2.4-1 libsoxr0
  libspatialite7 libspeex1 libssh-gcrypt-4 libsuperlu5 libswresample-dev
  libswresample3 libswscale-dev libswscale5 libsz2 libtbb-dev libtbb2
  libtesseract4 libthai-data libthai0 libtheora0 libtiff-dev libtiff5
  libtiffxx5 libtool libtwolame0 liburiparser1 libusb-1.0-0 libva-drm2
  libva-x11-2 libva2 libvdpau1 libvisual-0.4-0 libvorbis0a libvorbisenc2
  libvorbisfile3 libvpx6 libvtk6.3 libvulkan1 libwavpack1 libwayland-client0
  libwayland-cursor0 libwayland-egl1 libwebp6 libwebpmux3 libx11-xcb1
  libx264-155 libx265-179 libxcb-dri2-0 libxcb-dri3-0 libxcb-glx0
  libxcb-present0 libxcb-randr0 libxcb-render0 libxcb-shm0 libxcb-sync1
  libxcb-xfixes0 libxcomposite1 libxcursor1 libxdamage1 libxerces-c3.2
  libxfixes3 libxi6 libxinerama1 libxkbcommon0 libxnvctrl0 libxpm4 libxrandr2
  libxrender1 libxshmfence1 libxt6 libxtst6 libxvidcore4 libxxf86vm1
  libzvbi-common libzvbi0 m4 mesa-va-drivers mesa-vdpau-drivers
  mesa-vulkan-drivers networkd-dispatcher ocl-icd-libopencl1 odbcinst
  odbcinst1debian2 opencv-data openmpi-bin openmpi-common poppler-data
  proj-bin proj-data python3-dbus python3-gi shared-mime-info systemd
  systemd-sysv systemd-timesyncd ubuntu-mono ucf va-driver-all
  vdpau-driver-all x11-common xkb-data
```
These include multiple drivers and desktop libraries and require a total of 1200MB.