ros-perception/vision_opencv #158: Python3 compatibility

hgaiser, 2017-01-24:
Hey,

As is my understanding, ROS packages should aim to be [python2 and python3](http://wiki.ros.org/python_2_and_3_compatible_code) compatible. Currently on my system cv_bridge is linking against `libboost_python.so` though, which causes an issue for me since I am using a python3 interpreter and `libboost_python.so` is for python2 bindings (`libboost_python3.so` is for python3 bindings). The issue seems to come from [this](https://github.com/ros-perception/vision_opencv/blob/kinetic/cv_bridge/CMakeLists.txt#L7) line, which should be `find_package(Boost REQUIRED python3)` in my case.

However simply changing this to python3 will break python2 compatibility. Perhaps some option should be added to the CMakeLists.txt for choosing python3/python2.

Best regards,
Hans

vrabaud, 2017-01-29:
So, I solved it by checking which version of Python is installed. Now, as you mentioned, there should probably something at the level of CMake: what if you have Python2 installed and want to build for Python 3 ? (not sure we want to support that),

closed 2017-01-29 by vrabaud

hgaiser, 2017-01-30:
Yes I agree it would be better to have a boolean to determine if you want to compile with Python 2 or 3. I'm thinking it'd be better if it was a catkin flag, which is passed down to cmake. That way there'd be a ROS-wide flag of determining to compile with Python 2/3.

hdamron17, 2020-06-25:
Would it be possible to compile the version dependent part for both python 2 and 3 then use [`six`](https://pypi.org/project/six/) to determine at runtime which to import?

commit 5ae97283 (2017-01-29): properly find Boost Python 2 or 3
This fixes #158
commit 981656dd (2017-01-29): properly find Boost Python 2 or 3
This fixes #158