ros-perception/vision_opencv #161: Using tensorflow.contrib with cv_bridge causes tcmalloc error

lobachevzky, 2017-03-06:
### Environment info
Operating System:
```
❯ uname -a 
Linux dos 3.13.0-76-generic #120-Ubuntu SMP Mon Jan 18 15:59:10 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux
```

### Minimal reproducible example.
```
import tensorflow.contrib
import cv_bridge

import rospy
rospy.init_node('node')
```
This throws the following error:
```
/usr/bin/python2.7 /home/ethan/.PyCharmCE2016.3/config/scratches/scratch_4.py
src/tcmalloc.cc:277] Attempt to free invalid pointer 0xa2e78616d5f7475 

Process finished with exit code 134 (interrupted by signal 6: SIGABRT)
```
I've also posted to the tensorflow page (https://github.com/tensorflow/tensorflow/issues/8146)

### What other attempted solutions have you tried?
I tried reinstalling ros and tensorflow. No change. I also tried `print(cv_bridge.__file__)` to make sure I was importing the right directory for `cv_bridge`.

lobachevzky, 2017-03-08:
The solution was to switch the order of imports:
```
import tensorflow.contrib
import cv_bridge

import rospy
rospy.init_node('node')
```
Nevertheless, this is clearly a bug.

lobachevzky, 2017-03-09:
Ok. I did a little more digging and I found this line in my `.zshrc`:
```bash
export LD_PRELOAD="/usr/lib/libtcmalloc_minimal.so.4"  
```
So this line was causing the problem. I'm not really sure which of the three libraries that were involved would be responsible, but it might be good to include a more informative error message.

vrabaud, 2017-03-13:
ok, that is a tough one that really depends on your setting. What does libtcmalloc_minimal do ? Is OpenCV linked to it ?
Seems like it got fixed in tensorflow no ? https://github.com/tensorflow/tensorflow/issues/7960

closed 2017-03-13 by vrabaud