ros-perception/vision_opencv #162: cv_bridge needs rebuild for ROS Kinetic

roehling, 2017-03-17:
The recent upgrade of the opencv3 package to OpenCV 3.2 breaks the published cv_bridge package (which is still linked against OpenCV 3.1). I guess the ROS build farm failed to rebuild cv_bridge properly.

vrabaud, 2017-03-17:
It was re-built on the same day as planned:
http://build.ros.org/view/Kbin_uX64/job/Kbin_uX64__opencv3__ubuntu_xenial_amd64__binary/
http://build.ros.org/view/Kbin_uX64/job/Kbin_uX64__cv_bridge__ubuntu_xenial_amd64__binary/
What error do you have ? On the farm, do you have the logs ?
OpenCV got bumped from 3.1 to 3.2 right after its release (so end of December).

roehling, 2017-03-17:
I cleaned my workspace and rebuilt everything again from scratch, and it works now. I'm pretty sure I tried this before filing the issue, but never mind...

I'm not entirely sure why I did not run into this issue earlier, but my guess is that the sync from build farm to `packages.ros.org` has been delayed. Sorry for the noise.

closed 2017-03-17 by roehling

vrabaud, 2017-03-17:
no problem. Noise is always useful :)

ubuntuslave, 2017-03-17:
`cv_bridge` also seems to be confusing the linker for **ROS Indigo** which only supports `opencv 3.1` as per today. I had to disabled `cv_bridge` in order to use OpenCV 3.* without runtime errors. It works fine under **ROS Kinetic** though.