ros-perception/vision_opencv #169: CompressedImageMsg and alpha channels

laxnpander, 2017-05-29:
Hey there,

after some tests I found out that the compressed image message does not support an alpha channel. At least it is cutting them off and sets everything to 255 when decoding. Is this on purpose?

Best regards,

Alex

vrabaud, 2017-06-05:
what are you referring to here ? ImageTransport ? JPEG does not support alpha, PNG does.

laxnpander, 2017-06-08:
I have this code snippet as example:
`    // Converting input image "pImg" to ros msg
    sensor_msgs::ImagePtr imgMsg = cv_bridge::CvImage(header, "bgra8", pImg).toCompressedImageMsg(cv_bridge::PNG);
    // Converting ros msg back to cv::Mat
    cv_bridge::CvImagePtr imgPtr = cv_bridge::toCvCopy(imgMsg, "bgra8");
    cv::Mat img = imgPtr->image;
    
    std::cout << "Value pimg: " << pImg.at<cv::Vec4b>(0, 0) << std::endl;  // Outputs alpha 0
    std::cout << "Value img: " << img.at<cv::Vec4b>(0, 0) << std::endl;`  // Outputs alpha 255

My pImg is the input 4 channel cv::Mat that is converted to an bgra8 image message. Afterwards it is getting reconverted into a cv::Mat. Value at point (0, 0) is then shown for both input and reconverted mat. Alpha channel is lost and set to 255.  The same code works for "toImageMsg()".
Am I missing something?

vrabaud, 2017-06-08:
Thx for reporting, please confirm that master works for you

closed 2017-06-08 by vrabaud

laxnpander, 2017-06-09:
Confirmed

vrabaud, 2017-06-13:
Cool ! Yes, it got cherry-picked in Indigo: https://github.com/ros-perception/vision_opencv/commit/87b9bbaaac30379ca8888110657a177a33dbcc8a
I will make a release very soon: I am waiting for OpenCV 3.2.1 to be out this week and I'll release all of that together.

commit 8c659ac8 (2017-06-08): Properly deal with alpha in image compression.
That fixes #169
commit 87b9bbaa (2017-06-08): Properly deal with alpha in image compression.
That fixes #169