ros-perception/vision_opencv #170: Request Re-release

allenh1, 2017-06-05:
Hey, so there's a ticket that's been merged into OpenCV3 recently that is needed for compiling on Gentoo Linux: opencv/opencv#7985, merged in with opencv/opencv#7462.

Any way you could re-release?

Here's the relevant section of the cakin build.

```
==> make -j8 -l8 in '/var/tmp/portage/ros-lunar/opencv3-3.2.0-r6/work/build_isolated/opencv3/install'
Scanning dependencies of target opencv_core_pch_dephelp
Scanning dependencies of target opencv_cudev
Scanning dependencies of target opencv_cudaarithm_pch_dephelp
Scanning dependencies of target opencv_flann_pch_dephelp
Scanning dependencies of target opencv_hdf_pch_dephelp
Scanning dependencies of target opencv_imgproc_pch_dephelp
Scanning dependencies of target IlmImf
Scanning dependencies of target opencv_ml_pch_dephelp
[  0%] Building CXX object modules/cudev/CMakeFiles/opencv_cudev.dir/src/stub.cpp.o
[  0%] Building CXX object modules/hdf/CMakeFiles/opencv_hdf_pch_dephelp.dir/opencv_hdf_pch_dephelp.cxx.o
[  0%] Building CXX object modules/imgproc/CMakeFiles/opencv_imgproc_pch_dephelp.dir/opencv_imgproc_pch_dephelp.cxx.o
[  0%] Building CXX object modules/ml/CMakeFiles/opencv_ml_pch_dephelp.dir/opencv_ml_pch_dephelp.cxx.o
[  0%] Linking CXX shared library ../../lib/libopencv_cudev3.so
[  0%] Building CXX object modules/flann/CMakeFiles/opencv_flann_pch_dephelp.dir/opencv_flann_pch_dephelp.cxx.o
[  0%] Building CXX object modules/cudaarithm/CMakeFiles/opencv_cudaarithm_pch_dephelp.dir/opencv_cudaarithm_pch_dephelp.cxx.o
[  0%] Building CXX object modules/core/CMakeFiles/opencv_core_pch_dephelp.dir/opencv_core_pch_dephelp.cxx.o
[  0%] Built target opencv_cudev
Scanning dependencies of target opencv_reg_pch_dephelp
[  0%] Building CXX object modules/reg/CMakeFiles/opencv_reg_pch_dephelp.dir/opencv_reg_pch_dephelp.cxx.o
[  0%] Linking CXX static library ../../lib/libopencv_reg_pch_dephelp.a
[  0%] Built target opencv_reg_pch_dephelp
Scanning dependencies of target opencv_surface_matching_pch_dephelp
[  0%] Building CXX object 3rdparty/openexr/CMakeFiles/IlmImf.dir/Half/half.cpp.o
[  1%] Building CXX object modules/surface_matching/CMakeFiles/opencv_surface_matching_pch_dephelp.dir/opencv_surface_matching_pch_dephelp.cxx.o
[  1%] Linking CXX static library ../../lib/libopencv_hdf_pch_dephelp.a
[  1%] Built target opencv_hdf_pch_dephelp
Scanning dependencies of target opencv_video_pch_dephelp
[  1%] Building CXX object modules/video/CMakeFiles/opencv_video_pch_dephelp.dir/opencv_video_pch_dephelp.cxx.o
[  1%] Building CXX object 3rdparty/openexr/CMakeFiles/IlmImf.dir/Iex/IexBaseExc.cpp.o
[  1%] Linking CXX static library ../../lib/libopencv_cudaarithm_pch_dephelp.a
[  1%] Built target opencv_cudaarithm_pch_dephelp
Scanning dependencies of target opencv_viz_pch_dephelp
[  1%] Building CXX object modules/viz/CMakeFiles/opencv_viz_pch_dephelp.dir/opencv_viz_pch_dephelp.cxx.o
[  1%] Linking CXX static library ../../lib/libopencv_imgproc_pch_dephelp.a
[  2%] Linking CXX static library ../../lib/libopencv_ml_pch_dephelp.a
[  2%] Building CXX object 3rdparty/openexr/CMakeFiles/IlmImf.dir/Iex/IexThrowErrnoExc.cpp.o
[  2%] Built target opencv_imgproc_pch_dephelp
Scanning dependencies of target opencv_cudabgsegm_pch_dephelp
[  2%] Built target opencv_ml_pch_dephelp
Scanning dependencies of target opencv_cudafilters_pch_dephelp
[  2%] Building CXX object modules/cudabgsegm/CMakeFiles/opencv_cudabgsegm_pch_dephelp.dir/opencv_cudabgsegm_pch_dephelp.cxx.o
In file included from /var/tmp/portage/ros-lunar/opencv3-3.2.0-r6/work/src/opencv3-3.2.0/modules/core/include/opencv2/core/hal/intrin.hpp:293:0,
                 from /var/tmp/portage/ros-lunar/opencv3-3.2.0-r6/work/src/opencv3-3.2.0/modules/core/src/precomp.hpp:84,
                 from /var/tmp/portage/ros-lunar/opencv3-3.2.0-r6/work/build_isolated/opencv3/install/modules/core/opencv_core_pch_dephelp.cxx:1:
/var/tmp/portage/ros-lunar/opencv3-3.2.0-r6/work/src/opencv3-3.2.0/modules/core/include/opencv2/core/hal/intrin_sse.hpp: In function ‘cv::v_float32x4 cv::v_cvt_f32(const cv::v_float16x4&)’:
/var/tmp/portage/ros-lunar/opencv3-3.2.0-r6/work/src/opencv3-3.2.0/modules/core/include/opencv2/core/hal/intrin_sse.hpp:1721:42: error: ‘_mm_cvtph_ps’ was not declared in this scope
     return v_float32x4(_mm_cvtph_ps(a.val));
                                          ^
/var/tmp/portage/ros-lunar/opencv3-3.2.0-r6/work/src/opencv3-3.2.0/modules/core/include/opencv2/core/hal/intrin_sse.hpp: In function ‘cv::v_float16x4 cv::v_cvt_f16(const cv::v_float32x4&)’:
/var/tmp/portage/ros-lunar/opencv3-3.2.0-r6/work/src/opencv3-3.2.0/modules/core/include/opencv2/core/hal/intrin_sse.hpp:1726:45: error: ‘_mm_cvtps_ph’ was not declared in this scope
     return v_float16x4(_mm_cvtps_ph(a.val, 0));
                                             ^
[  2%] Linking CXX static library ../../lib/libopencv_surface_matching_pch_dephelp.a
[  2%] Building CXX object modules/cudafilters/CMakeFiles/opencv_cudafilters_pch_dephelp.dir/opencv_cudafilters_pch_dephelp.cxx.o
[  2%] Built target opencv_surface_matching_pch_dephelp
[  3%] Linking CXX static library ../../lib/libopencv_flann_pch_dephelp.a
Scanning dependencies of target opencv_cudaimgproc_pch_dephelp
[  3%] Built target opencv_flann_pch_dephelp
Scanning dependencies of target opencv_cudawarping_pch_dephelp
make[2]: *** [modules/core/CMakeFiles/opencv_core_pch_dephelp.dir/build.make:63: modules/core/CMakeFiles/opencv_core_pch_dephelp.dir/opencv_core_pch_dephelp.cxx.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1939: modules/core/CMakeFiles/opencv_core_pch_dephelp.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
```

vrabaud, 2017-06-05:
Done for jade-lunar. Do we also need it for Indigo ? Just because I'd like to stay on 3.1.0 there.

vrabaud, 2017-06-05:
BTW, thx for reporting, that helps make ROS better !

closed 2017-06-05 by vrabaud

allenh1, 2017-06-05:
Doing it for indigo would be great, so long as it doesn't break things on your end. Thank you very much for doing this so quickly!

vrabaud, 2017-06-05:
ok, done for Indigo. It should be fine: it is released as 3.1 but it is actually the same version as the other ROS distros.

allenh1, 2017-06-05:
Thank you very much!