ros-perception/vision_opencv #173: compilation error after latest commits

klepsydra-robotics, 2017-06-18:
Dear all,

after the last commits of 8 and 10 June 2017, I can get image_geometry to compile. The error is:

```
/home/alerion/thirdparties/catkin_ws/src/vision_opencv/image_geometry/src/pinhole_camera_model.cpp: In member function ‘bool image_geometry::PinholeCameraModel::fromCameraInfo(const CameraInfo&)’:
/home/alerion/thirdparties/catkin_ws/src/vision_opencv/image_geometry/src/pinhole_camera_model.cpp:87:12: error: no match for ‘operator=’ (operand types are ‘boost::shared_ptr<image_geometry::PinholeCameraModel::Cache>’ and ‘std::shared_ptr<image_geometry::PinholeCameraModel::Cache>’)
     cache_ = std::make_shared<Cache>();
            ^
In file included from /usr/include/boost/shared_ptr.hpp:17:0,
                 from /usr/include/boost/format/alt_sstream.hpp:21,
                 from /usr/include/boost/format/internals.hpp:23,
                 from /usr/include/boost/format.hpp:38,
                 from /usr/include/boost/math/policies/error_handling.hpp:31,
                 from /usr/include/boost/math/special_functions/round.hpp:14,
                 from /opt/ros/kinetic/include/ros/time.h:58,
                 from /opt/ros/kinetic/include/ros/serialization.h:34,
                 from /opt/ros/kinetic/include/sensor_msgs/CameraInfo.h:14,
                 from /opt/ros/kinetic/include/image_geometry/pinhole_camera_model.h:4,
                 from /home/alerion/thirdparties/catkin_ws/src/vision_opencv/image_geometry/src/pinhole_camera_model.cpp:3:
/usr/include/boost/smart_ptr/shared_ptr.hpp:500:18: note: candidate: boost::shared_ptr<T>& boost::shared_ptr<T>::operator=(const boost::shared_ptr<T>&) [with T = image_geometry::PinholeCameraModel::Cache]
     shared_ptr & operator=( shared_ptr const & r ) BOOST_NOEXCEPT
```

Attached is the full error as well.
[compilation_error.txt](https://github.com/ros-perception/vision_opencv/files/1082909/compilation_error.txt)

Thanks,

Pablo

vrabaud, 2017-06-19:
Right, we changed the ABI and that's bad (and that creates that bug),
Nonetheless, your gcc includes /opt/ros/kinetic/include/image_geometry/pinhole_camera_model.h and it should include the one from your workspace. And that's not valid.
I don't know what you did with your workspace so you should delete it and rebuild.

closed 2017-06-19 by vrabaud

evenator, 2017-07-08:
I think the cause of this error is actually the order of the `include_directories()` statements in image_geometry/CMakeLists.txt. Because the package local include directory is listed *after* the `${catkin_LIBRARIES}`, this error will manifest on any system with the old headers installed to `/opt/ros`. I recommend reordering the include directories in the CMakeLists.txt file so that the package local headers take precedence over the system headers.