ros-perception/vision_opencv #177: [cv_bridge][indigo] imgmsg_to_cv2 returns ndim=3 ndarray even when desired_encoding='mono8' with opencv2

knorth55, 2017-07-06:
ROS: indigo
Ubuntu: 14.04
OpenCV: OpenCV2

`imgmsg_to_cv2` returns `ndim=3` `ndarray` even when `desired_encoding='mono8'` with OpenCV2.
I looked and tried to fix this issue.
In this line https://github.com/ros-perception/vision_opencv/blob/indigo/cv_bridge/python/cv_bridge/core.py#L182, 

```python
 res = cvtColor2(im, img_msg.encoding, desired_encoding)
```
`im.ndim` is `2` but `res.ndim` is `3` when you set `desired_encoding='mono8'`.

knorth55, 2017-07-06:
I checked this does not happen with OpenCV3.

vrabaud, 2017-07-17:
hi, could you please write a test for that ? Did that get fixed in OpenCV upstream and module_opencv2.cpp just needs to be updated ?

knorth55, 2017-07-18:
You can check with code below.
```
#!/usr/bin/env python

import cv_bridge
import numpy as np

bridge = cv_bridge.CvBridge()
mono = np.random.random((100, 100)) * 255
mono = mono.astype(np.uint8)

input_msg = bridge.cv2_to_imgmsg(mono, encoding='mono8')
print('input_msg height  : {}'.format(input_msg.height))
print('input_msg width   : {}'.format(input_msg.width))
print('input_msg step    : {}'.format(input_msg.step))
print('input_msg encoding: {}'.format(input_msg.encoding))
output = bridge.imgmsg_to_cv2(input_msg, desired_encoding='mono8')
print('>>>>>>>>>>>>>>>>>>>>>>')
print('output dtype      : {}'.format(output.dtype))
print('output shape      : {}'.format(output.shape))
```
output is below.

```
input_msg height  : 100
input_msg width   : 100
input_msg step    : 100
input_msg encoding: mono8
>>>>>>>>>>>>>>>>>>>>>>
output dtype      : uint8
output shape      : (100, 100, 1)
```

knorth55, 2017-07-18:
>Did that get fixed in OpenCV upstream and module_opencv2.cpp just needs to be updated ?

sorry, but I do not follow OpenCV upstream changes.
let me take a look.

closed 2017-08-06 by vrabaud

knorth55, 2017-08-07:
thank you for solving this issue!

Alex-Beh, 2020-06-13:
@knorth55 Hello, may I know the version of the opencv2 that you use? I try use opencv 2.4.9 to compile the melodic branch and face the following cmake error when I catkin_make:
```
CMake Error at /usr/local/share/OpenCV/OpenCVConfig.cmake:205 (message):
  opencv_imgcodecs is required but was not found
```
It seems the imgcodecs module is not present in earlier versions of OpenCV (e.g. 2.4.9). .([url](https://stackoverflow.com/questions/25195031/opencv-imgcodecs-hpp-file-not-found-error))

knorth55, 2020-06-13:
@Alex-Beh please check here. https://www.ros.org/reps/rep-0003.html#melodic-morenia-may-2018-may-2023
OpenCV 3.2 is targeted for melodic.

commit d19a5886 (2017-08-06): Fixes #177

The Python bridge was wrong on OpenCV2 with mono8 (and any Mat
with only two dimensions btw). Took the official Python bridge
from OpenCV.
commit 3695c319 (2017-08-06): Fixes #177

The Python bridge was wrong on OpenCV2 with mono8 (and any Mat
with only two dimensions btw). Took the official Python bridge
from OpenCV.