ros-perception/vision_opencv #182: PinholeCameraModel() doesn't import in Python

SimonBirrell, 2017-07-12:
Ubuntu 16.04, Kinetic installation.

Trying to import the PinHoleCamera() model in Python, using:

    from image_geometry import PinholeCameraModel

This gives an error:

```
Traceback (most recent call last):
  File "/home/simonbirrell/catkin_ws/src/vegebot/lettuce_detect/nodes/lettuce_detector", line 16, in <module>
    from image_geometry import PinholeCameraModel
ImportError: cannot import name PinholeCameraModel
```

This guy appears to have the same problem:
http://answers.ros.org/question/209953/image_geometry-pinholecameramodel-python-not-importing-properly-when-installed-using-debian-package/

Following his suggestion I tried putting the image_geometry package in catkin_ws/src and compiling with catkin_make, but got the following:    

```
CMake Error at /home/simonbirrell/catkin_ws/devel/share/image_geometry/cmake/image_geometryConfig.cmake:106 (message):
  Project 'image_geometry' specifies
  '/home/simonbirrell/catkin_ws/src/image_geometry/include' as an include
  dir, which is not found.  It does neither exist as an absolute directory
  nor in
  '/home/simonbirrell/catkin_ws/src/image_geometry//home/simonbirrell/catkin_ws/src/image_geometry/include'.
  Ask the maintainer 'Vincent Rabaud <vincent.rabaud@gmail.com>' to fix it.
Call Stack (most recent call first):
  /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:76 (find_package)
  vegebot/lettuce_detect/CMakeLists.txt:7 (find_package)

-- Configuring incomplete, errors occurred!
```

Any thoughts on a workaround or proper fix?

Thanks

Simon

vrabaud, 2017-07-14:
Can you please try with the latest from source from cv_bridge ? Thx

SimonBirrell, 2017-07-17:
That works! Thank you!

Simon

vrabaud, 2017-11-05:
Releasing the fix.

closed 2017-11-05 by vrabaud