ros-perception/vision_opencv #192: cv_bridge moving from opencv 3.2 to 3.3 no longer links libopencv_highgui3

lucasw, 2017-12-11:
I just did an apt upgrade in kinetic and now C++ that used to get highgui via cv_bridge no longer links.

mikaelarguedas, 2017-12-11:
Yes indeed, now cv_bridge only links against what it needs (https://github.com/ros-perception/vision_opencv/commit/8b5bbcbc1ce65734dc600695487909e0c67c1033).
Packages using openCV should `find_package` and link against the components they use.

AFAIK all released packages in Kinetic and Lunar have been updated accordingly and re-released. If you find any package that don't build because of this,  please consider open an issue or consider opening a PR with the fix.

Thanks!

lucasw, 2017-12-11:
It's just all my packages that need to be updated (and P.J. Reed's https://discourse.ros.org/t/opencv-3-3-1-is-breaking-builds-in-kinetic/3384).

This is what is now required to get highgui back:

```
find_package(OpenCV REQUIRED)
...
target_link_libraries(
  ...
  ${OpenCV_LIBRARIES}
```

A note in http://wiki.ros.org/opencv3 would be good (I can do it later).

tfoote, 2017-12-12:
@lucasw Can you try adding OpenCV as a catkin_project `DEPENDS` https://github.com/lucasw/vimjay/commit/45daf25a802fc199bb021aad049350ad3795d212#diff-af3b638bc2a3e6c650974192a53c7291R105 Like here: https://github.com/swri-robotics/marti_common/blob/master/swri_geometry_util/CMakeLists.txt#L36

tfoote, 2018-03-16:
Is there anything else that needs to be follow up on this. @lucasw Did you get a chance to make a note on the wiki?

closed 2018-03-20 by tfoote