ros-perception/vision_opencv #193: target_link_libraries with ${OpenCV_LIBRARIES} fails with 3.3.1

tfoote, 2017-12-12:
There's a reported regression relating to linking against OpenCV 3.3.1 in this [ROS Discourse thread](https://discourse.ros.org/t/opencv-3-3-1-is-breaking-builds-in-kinetic/3384?u=tfoote) also releated #192

pjreed, 2017-12-12:
Thanks.  The workaround suggested in the thread looks like it works, although it's not ideal since it requires modifying package build files, and I don't normally bother declaring `DEPENDS` in the `catkin_package` macro for packages that don't export any libraries.

Another workaround I found is to modify the OpenCVConfig.cmake file to return absolute paths to its libraries, like so:

```bash
sed -i -r \
  's#set\(OpenCV_LIBS \$\{OpenCV_LIBS\} "\$\{__cvcomponent\}"\)#set\(OpenCV_LIBS \$\{OpenCV_LIBS\} "\$\{OpenCV_INSTALL_PATH\}/lib/lib\$\{__cvcomponent\}3.so")#' \
  /opt/ros/kinetic/share/OpenCV-3.3.1/OpenCVConfig.cmake
```

But also not ideal since that involves modifying a system-installed file.

tdenewiler, 2017-12-12:
I am working with my [camera driver](https://github.com/tdenewiler/foscam_8918_driver) that uses OpenCV to make a connection to a web camera. When I add `DEPENDS OpenCV` and remove `${OpenCV_LIBRARIES}` from the `target_link_libraries` command I get linker errors such as

    /src/foscam_8918_driver/src/foscam_8918_driver/src/foscam_8918_driver.cpp:55: undefined reference to `cv::VideoCapture::open(cv::String const&)'

tdenewiler, 2017-12-12:
I was able to build successfully by explicitly calling out the OpenCV components I use, as in [this commit](https://github.com/tdenewiler/foscam_8918_driver/commit/c49b9a0c87645977cee1b6f6c4a03324f0df3df8).

tdenewiler, 2017-12-12:
Using the https://github.com/tdenewiler/foscam_8918_driver package as the only package in a fresh workspace I get successful builds when using `catkin_make` or `catkin_make -DCMAKE_BUILD_TYPE=Release`.

However, from a clean workspace when I use `catkin_make -DCMAKE_BUILD_TYPE=RelWithDebInfo` I get the original error of `make[2]: *** No rule to make target 'opencv_core-NOTFOUND`.

I get the same results when substituting `catkin build` for `catkin_make`.

pjreed, 2017-12-12:
Interesting; comparing the diffs between the OpenCVConfig.cmake files from 3.2.0 and 3.3.1, I see that the 3.3.1 version adds this block at line 204:

```cmake
  if(TARGET ${__cvcomponent})
      set_target_properties(${__cvcomponent} PROPERTIES
          MAP_IMPORTED_CONFIG_MINSIZEREL "Release"
          MAP_IMPORTED_CONFIG_RELWITHDEBINFO "Release"
      )
  endif()
```

If I comment out that block, my build with `RelWithDebInfo` set passes, but I'm not sure why that would affect it.

tdenewiler, 2017-12-12:
I verified that commenting out the `if()` block (line 204) in `/opt/ros/kinetic/share/OpenCV-3.3.1/OpenCVConfig.cmake` does allow my package to build in `RelWithDebInfo` mode.

tfoote, 2017-12-12:
It looks like this came from https://github.com/opencv/opencv/commit/2d0611bc01b7c51bff8567c332077bc5c3f64c10 https://github.com/opencv/opencv/pull/9656

Where thie issue with Kinetic was flagged yesterday too.

pbeeson, 2017-12-13:
I can verify that removing the OpenCV_LIBRARIES from targets (as advocated here: https://discourse.ros.org/t/opencv-3-3-1-is-breaking-builds-in-kinetic/3384/5) does not work for all cases.  Some functions are now missing critical libraries.
undefined reference to `cv::findTransformECC...`

tfoote, 2017-12-13:
For anyone finding this there's now a proposed fix in OpenCV: https://github.com/opencv/opencv/pull/10299. If you're running into this please try the patch and confirm with them upstream.

asymingt, 2017-12-13:
I'm not proud of it, but you can use this interim fix to discover and prepend the absolute paths to the OpenCV components, so you don't have to overwrite the ```OpenCVConfig.cmake`` system file:

```cmake
# Find OpenCV
find_package(OpenCV QUIET NO_MODULE)

# Get the library suffix
set(CVLIB_SUFFIX ".so.${OpenCV_VERSION_MAJOR}.${OpenCV_VERSION_MINOR}.${OpenCV_VERSION_PATCH}")

# Loop over each component adding the absolute path
set(OpenCV_LIBRARIES "")
foreach(__CVLIB ${OpenCV_LIBS})
  find_library(OpenCV_${__CVLIB}_LIBRARY
    "${__CVLIB}3${CVLIB_SUFFIX}"
    "${__CVLIB}3${CVLIB_SUFFIX}d"
    "lib${__CVLIB}3${CVLIB_SUFFIX}"
    "lib${__CVLIB}3${CVLIB_SUFFIX}d"
    PATHS  "${OpenCV_DIR}/../../lib" NO_DEFAULT_PATH)
  list(APPEND OpenCV_LIBRARIES ${OpenCV_${__CVLIB}_LIBRARY})
endforeach(__CVLIB)

# For those who use the OpenCV_LIBS flag
set(OpenCV_LIBS ${OpenCV_LIBRARIES})
```

asymingt, 2017-12-13:
While we wait for a fix, this seems to fix the issue without requiring a system file change.

```cmake
find_package(OpenCV 3.3.1 REQUIRED)
foreach(__cvcomponent ${OpenCV_LIB_COMPONENTS})
  set (__original_cvcomponent ${__cvcomponent})
  if(NOT __cvcomponent MATCHES "^opencv_")
    set(__cvcomponent opencv_${__cvcomponent})
  endif()
  if (TARGET ${__cvcomponent})
    set_target_properties(${__cvcomponent} PROPERTIES
        MAP_IMPORTED_CONFIG_DEBUG ""
        MAP_IMPORTED_CONFIG_RELEASE ""
        MAP_IMPORTED_CONFIG_RELWITHDEBINFO ""
        MAP_IMPORTED_CONFIG_MINSIZEREL ""
    )
  endif()
endforeach(__cvcomponent)
set(OpenCV_LIBRARIES ${OpenCV_LIBS})
```

tfoote, 2017-12-27:
OpenCV 3.4 came out with this resolved: https://github.com/opencv/opencv/releases/tag/3.4.0 @vrabaud would you have a moment to review and push the new released version?

Triocrossing, 2018-01-06:
I ran into this problem and finally fix it by erase and rebuild my proj in the working space, someone says it links with the pre-build problem.

pbeeson, 2018-01-12:
Is the upgrade to OpenCV 3.4.0 being investigated?

pbeeson, 2018-01-12:
Or can we perhaps push out a 3.3.1-1 patch that just fixes the CMake file using one of the solutions above?

keenan-burnett, 2018-01-15:
Experienced this issue today (Ubuntu 16.04 + ROS Kinetic) while attempting to build a ROS package which depends on OpenCV. I had just installed OpenCV 3.3.1 from source. Commenting out the if statement at line 204 in /opt/ros/kinetic/share/OpenCV-3.3.1/OpenCVConfig.cmake worked for me.

tfoote, 2018-03-16:
We've updated the build of opencv3 to use Release: https://github.com/ros-gbp/opencv3-release/pull/15 re https://github.com/ros-gbp/opencv3-release/issues/14 

It's in the testing repository now. Please see if that resolve these issues.

pauldinh, 2018-03-19:
I'm having success with the deb pulled from the shadow repo. Looks good to me, thanks!

tfoote, 2018-03-20:
Thanks for validating I believe that this is resolved enough to close.

closed 2018-03-20 by tfoote