ros-perception/vision_opencv #196: Unable to import cv2 and cvbridge in python3

zishanahmed08, 2017-12-17:
**Environment**
Ubuntu 16.04
Python 3.5.2
Ros Distribution-Kinetic

Ros was installed as per installation instruction (sudo apt-get install ros-kinetic-full-desktop.)
And to make the Python3 as default version, I followed the necessary and sufficient step of install python3 versions of rospkg and catkin(sudo pip3 install rospkg catkin_pkg).After which i am able to import cv2 in python2 but  not in python3 as below
**ImportError: /opt/ros/kinetic/lib/python2.7/dist-packages/cv2.so: undefined symbol: PyCObject_Type**

To solve this I install the opencv3 with (pip3 install opencv-python) and then force a symbolic link as mentioned [here](https://stackoverflow.com/questions/43019951/after-install-ros-kinetic-cannot-import-opencv)
But then i am unable to import cv_bridge

So whats the right way to install ROS such that the python version is python3 and am able to import opencv and corresponding dependencies

oneway234, 2018-04-16:
Hi,
I'm using the same environment. 
Ubuntu 16.04
Python 3.5.2
Ros Distribution-Kinetic

I had been disturbing by this problem for these days. And finally I found a solution to fix this problem. As we added this line "source /opt/ros/kinetic/setup.bash" in file "~/.bashrc", our ubuntu will work in ROS environment. And it will result this issue, because it seems like ROS will only create folder "python2.7", so when we using python3 it cannot successfully import cv2. 

**So, what we need to do is go to /opt/ros/kinetic and rename the folder "python2.7" to "python3.5". Then it can work.**

walid19, 2018-08-31:
I am having the same problem did you find any solution?

saideepthik, 2018-12-26:
i'm also having same problem 
and i installed opencv in python3 also still getting same import error don't know how to resolve????

tykurtz, 2019-02-27:
> @saideepthik [this](https://stackoverflow.com/a/52250239) answer usually works the best. Note that you might have to use sudo while removing the file. It works for me!

That answer worked for importing cv2 in python3, but cv_bridge was still broken.

[Building from source](https://stackoverflow.com/a/50291787) worked out well for me to get cv_bridge running under python3. Only difference from the author was that I made a symlink `ln -s /usr/lib/x86_64-linux-gnu/libboost_python-py35.so /usr/lib/x86_64-linux-gnu/libboost_python3.so` instead of modifying the CMake file.

fnando1995, 2019-06-15:
I am having the same problem.

SYSTEM: Ubuntu 16.04/ROS Kinetic/python3.5

I manage to use the answer of @tykurtz here, using the symlink with the other stackoverflow answer. and when using the cv_bridge import works fine (i did not check the functions).

but, as the stackoverflow answer says, it compiles using catkin build, i am used to use ctkin_make command instead. so i erase the build & devel folders at the workspace and make the catkin_make and compile went well, also if i try (see code), python still can compile the cv_bridge import.

```
user@machine:~/catkin$ python3
Python 3.5.2 (default, Nov 12 2018, 13:43:14) 
[GCC 5.4.0 20160609] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from cv_bridge.boost.cv_bridge_boost import getCvType
>>>
```
I was happy for this, but when trying to keep my project going, the services where not build as expected... any help here?

avinashsen707, 2020-02-03:
You can simple close the problem by exporting the path to python3.

1. Echo the current python path
2. Unset the current python path
3. Export the path to python3

fnando1995, 2020-02-03:
@avinashsen707
that's actually the solution for the problem. did it long time ago. Thanks for reminder this thread thought.!

Hiroaki-K4, 2020-12-24:
Hi @avinashsen707 @fnando1995 .
I have a same problem. Can you tell me the specific command executed in each step?

fnando1995, 2020-12-24:
@Hiroaki-K4 edit the python library path in the environment variable.  Else, in code, you can use import sys; sys.path.append("path to python 3"). See that this command "appends" a path to the python library path during the execution of the code.

Hiroaki-K4, 2020-12-28:
@fnando1995 Thank you for reply.
I used this [link](https://stackoverflow.com/questions/49221565/unable-to-use-cv-bridge-with-ros-kinetic-and-python3/50291787#50291787) as a reference to resolve the error regarding cv_bridge.
Here is the result of executing the command before trying the link method.

```
(jetson) nvidia@nvidia:~/catkin_ws$ python
Python 3.6.9 (default, Oct  8 2020, 12:12:24) 
[GCC 8.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import sys
>>> import pprint
>>> pprint.pprint(sys.path)
['',
 '/opt/ros/melodic/lib/python2.7/dist-packages',
 '/home/nvidia/catkin_ws/devel/lib/python2.7/dist-packages',
 '/usr/local/lib',
 '/home/nvidia/virtualenv/jetson/lib/python36.zip',
 '/home/nvidia/virtualenv/jetson/lib/python3.6',
 '/home/nvidia/virtualenv/jetson/lib/python3.6/lib-dynload',
 '/usr/lib/python3.6',
 '/home/nvidia/virtualenv/jetson/lib/python3.6/site-packages',
 '/home/nvidia/virtualenv/jetson/lib/python3.6/site-packages/trt_pose-0.0.1-py3.6-linux-aarch64.egg']
```
Here is the result of executing the command after trying the link method.

```
(jetson) nvidia@nvidia:~/catkin_ws$ python
Python 3.6.9 (default, Oct  8 2020, 12:12:24) 
[GCC 8.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import sys
>>> import pprint
>>> pprint.pprint(sys.path)
['',
 '/home/nvidia/catkin_ws/devel/lib/python3/dist-packages',
 '/opt/ros/melodic/lib/python2.7/dist-packages',
 '/home/nvidia/catkin_ws/devel/lib/python2.7/dist-packages',
 '/usr/local/lib',
 '/home/nvidia/virtualenv/jetson/lib/python36.zip',
 '/home/nvidia/virtualenv/jetson/lib/python3.6',
 '/home/nvidia/virtualenv/jetson/lib/python3.6/lib-dynload',
 '/usr/lib/python3.6',
 '/home/nvidia/virtualenv/jetson/lib/python3.6/site-packages',
 '/home/nvidia/virtualenv/jetson/lib/python3.6/site-packages/trt_pose-0.0.1-py3.6-linux-aarch64.egg']
```
You say I should add the python library path using appends, but does the python library path mean -DPYTHON_LIBRARY = / usr / lib / x86_64-linux-gnu / libpython3.5m.so for [link](https://stackoverflow.com/questions/49221565/unable-to-use-cv-bridge-with-ros-kinetic-and-python3/50291787#50291787)?
Thanks!

ijnek, 2022-09-12:
I recently became the maintainer of this repo, and I'm sorting through issues, closing stale ones. As this issue is raised against a distro that has reached EOL, I'm going to close it out. Please feel free to comment if it is still relevant in other active distros and you would like to get it re-opened.

closed 2022-09-12 by ijnek