ros-perception/vision_opencv #197: OpenCV cuda parts missing

Dominik-FKIE, 2017-12-18:
Thanks a lot for packaging OpenCV, I appreciate your work very much!

My problem is that cuda accelerated parts of OpenCV, namely cudaarithm, cudabgsegm, cudacodec, cudafeatures2d, cudafilters, cudaimgproc, cudalegacy, cudaobjdetect, cudaoptflow, cudastereo, cudawarping, and cudev are not included.

Standart build option of OpenCV is "WITH_CUDA" flag enabled and they are all part of the "main modules" according to documentation. So I was wondering if this absence is intended for some reason? If not, would you please include them in your future builds/releases?

corot, 2018-05-23:
I wonder the same

TomShkurti, 2020-04-03:
Bumped/agreed, having a ros-[whatever]-opencv3-CUDA package alongside ros-[whatever]-opencv3 would save my lab a lot of trouble. Would this just be the addition of another OpenCV base to the build process?

closed 2025-12-29 by clalancette