ros-perception/vision_opencv #201: no calib3d in cv_bridge libs

ClementLeBihan, 2018-01-22:
Hi,
When I add cv_bridge in my CMakeList, opencv_calib3d is absent from catkin_libraries.

Here is some command I typed to debug : 
`rospack libs-only-l cv_bridge 
cv_bridge rosconsole rosconsole_log4cxx rosconsole_backend_interface roscpp_serialization rostime cpp_common`

`rospack libs-only-other cv_bridge 
/opt/ros/kinetic/lib/libopencv_core3.so.3.3.1 /opt/ros/kinetic/lib/libopencv_imgproc3.so.3.3.1 /opt/ros/kinetic/lib/libopencv_imgcodecs3.so.3.3.1 /usr/lib/x86_64-linux-gnu/liblog4cxx.so /usr/lib/x86_64-linux-gnu/libboost_regex.so /usr/lib/x86_64-linux-gnu/libboost_system.so /usr/lib/x86_64-linux-gnu/libboost_thread.so /usr/lib/x86_64-linux-gnu/libboost_chrono.so /usr/lib/x86_64-linux-gnu/libboost_date_time.so /usr/lib/x86_64-linux-gnu/libboost_atomic.so /usr/lib/x86_64-linux-gnu/libpthread.so /usr/lib/x86_64-linux-gnu/libboost_date_time.so /usr/lib/x86_64-linux-gnu/libboost_system.so /usr/lib/x86_64-linux-gnu/libboost_thread.so /usr/lib/x86_64-linux-gnu/libboost_chrono.so /usr/lib/x86_64-linux-gnu/libboost_atomic.so /usr/lib/x86_64-linux-gnu/libpthread.so /usr/lib/x86_64-linux-gnu/libboost_system.so /usr/lib/x86_64-linux-gnu/libboost_thread.so /usr/lib/x86_64-linux-gnu/libboost_chrono.so /usr/lib/x86_64-linux-gnu/libboost_date_time.so /usr/lib/x86_64-linux-gnu/libboost_atomic.so /usr/lib/x86_64-linux-gnu/libpthread.so /usr/lib/x86_64-linux-gnu/libconsole_bridge.so`

`ls /opt/ros/kinetic/lib/ | grep calib3d
libopencv_calib3d3.so
libopencv_calib3d3.so.3.3
libopencv_calib3d3.so.3.3.1`

Does anyone can help me ?

meyerj, 2018-02-05:
It's on purpose, at least since https://github.com/ros-perception/vision_opencv/commit/8b5bbcbc1ce65734dc600695487909e0c67c1033. cv_bridge only exports the `opencv_core`, `opencv_imgproc` and `opencv_imgcodecs` modules because that's what is needed to provide it's functionality:
```
find_package(OpenCV 3 REQUIRED
  COMPONENTS
    opencv_core
    opencv_imgproc
    opencv_imgcodecs
  CONFIG
)
```
(from https://github.com/ros-perception/vision_opencv/blob/kinetic/cv_bridge/CMakeLists.txt#L16)

In contrast, the [image_geometry](http://wiki.ros.org/image_geometry) package exposes all available OpenCV modules, check https://github.com/ros-perception/vision_opencv/blob/kinetic/image_geometry/CMakeLists.txt#L5). But depending on that package also introduces a dependency to `sensor_msgs`.

The pros and cons of "misusing" one of both as a platform-independent way to find OpenCV are briefly described at http://wiki.ros.org/opencv3#package.xml.

If you want to have full control over what OpenCV libraries you link against, don't depend on any of those two and find a version of OpenCV yourself, using [find_package(OpenCV ...)](https://github.com/opencv/opencv/blob/master/cmake/templates/OpenCVConfig.cmake.in).

Or you could use a combination of `cv_bridge` and `find_package(OpenCV)` to find additional modules of the exact same OpenCV version used to build `cv_bridge`:
```cmake
# sets OpenCV_VERSION, among others...
find_package(catkin REQUIRED COMPONENTS cv_bridge ...)

# find additional OpenCV modules
find_package(OpenCV  ${OpenCV_VERSION} EXACT REQUIRED opencv_calib3d)

add_library(mytarget ...)
target_link_libraries(mytarget ${catkin_LIBRARIES} ${OpenCV_LIBS})
```

vrabaud, 2018-04-16:
Thx @meyerj  for the proper explanation

closed 2018-04-16 by vrabaud