ros-perception/vision_opencv #204: Problem with std::shared_ptr and std::make_shared

hz658832, 2018-03-18:
Hey there,

I found a small bug in the package image_geometry, where I got the  `error: ‘shared_ptr’ in namespace ‘std’ does not name a template type`.

And the solution for that could be replacing the std::shared_ptr and std::make_shared with boost::shared_ptr and boost::make_shared by adding the header for shared_ptr and make_shared also the find_package in CMakeList.

I do not think that we always get errors with STL, but perhaps boost can make the code more robust.
:)

mikaelarguedas, 2018-03-18:
Can you provide more information  about the platform and the way you build your packages?

It looks like [this commit](https://github.com/ros-perception/vision_opencv/commit/0b36ea7a6f40fc05d54e3e6871933129687d5cc4) provided the tick tock to use either boost or std for shared_ptr. Maybe an `#include <memory>` is missing somewhere the STL version is being used.

Before that this code was using only boost pointers but several ROS packages are trying to move away from boost dependencies now that the features used are available in STL.

hz658832, 2018-03-19:
Actually I added the memory header before I changed the std to boost. It didn’t help and I gave up to find out why😊.
The platform I am using is the NVIDIA Jetson TX2 with newest os and I just did the normal way with catkin_make.

I open this issue only want to give the information if someone has the same problem and I do think that std::* should also do the right job.

Thx.

mikaelarguedas, 2018-03-19:
So you are using Ubuntu Xenial on an arm64 processor and using ROS Kinetic (or ROS Lunar maybe?) ?
If you can provide the complete information of your system setup as well as how to reproduce the issue and the complete error message you get, that would be helpful for reproducing / investigating the issue.
Thanks

ijnek, 2022-09-12:
I recently became the maintainer of this repo, and I'm sorting through issues, closing stale ones. As this issue has not been active for a while (and has not gained support from other users), I'm going to close it. Please feel free to comment if it is still relevant and you would like to get it re-opened.

closed 2022-09-12 by ijnek