ros-perception/vision_opencv #205: can't process package cv_bridge

stefmorren, 2018-03-27:
I'm having an issue with my catkin_make.  I receive multiple errors after 
`Building CXX object src/CMakeFiles/cv_bridge_boostdir/module.cpp.o`

Errors:
```
In file included from /usr/include/c++/4.9/cstddef:45:0,
                 from /usr/include/boost/config/select_stdlib_config.hpp:18,
                 from /usr/include/boost/config.hpp:40,
                 from /usr/include/boost/python/detail/config.hpp:15,
                 from /usr/include/boost/python/detail/prefix.hpp:14,
                 from /usr/include/boost/python/args.hpp:8,
                 from /usr/include/boost/python.hpp:11,
                 from /home/robot/ros_catkin_ws/src/vision_opencv/cv_bridge/src/module.hpp:21,
                 from /home/robot/ros_catkin_ws/src/vision_opencv/cv_bridge/src/module.cpp:35:
/home/robot/ros_catkin_ws/src/vision_opencv/cv_bridge/src/module.hpp: In function 'void do_numpy_import()':
/usr/lib/python2.7/dist-packages/numpy/core/include/numpy/__multiarray_api.h:1708:35: error: return-statement with a value, in function returning 'void' [-fpermissive]
 #define NUMPY_IMPORT_ARRAY_RETVAL NULL
                                   ^
/usr/lib/python2.7/dist-packages/numpy/core/include/numpy/__multiarray_api.h:1713:151: note: in expansion of macro 'NUMPY_IMPORT_ARRAY_RETVAL'
 #define import_array() {if (_import_array() < 0) {PyErr_Print(); PyErr_SetString(PyExc_ImportError, "numpy.core.multiarray failed to import"); return NUMPY_IMPORT_ARRAY_RETVAL; } }
                                                                                                                                                       ^
/home/robot/ros_catkin_ws/src/vision_opencv/cv_bridge/src/module.hpp:44:5: note: in expansion of macro 'import_array'
     import_array( );
     ^
src/CMakeFiles/cv_bridge_boost.dir/build.make:54: recipe for target 'src/CMakeFiles/cv_bridge_boost.dir/module.cpp.o' failed
make[2]: *** [src/CMakeFiles/cv_bridge_boost.dir/module.cpp.o] Error 1
CMakeFiles/Makefile2:918: recipe for target 'src/CMakeFiles/cv_bridge_boost.dir/all' failed
make[1]: *** [src/CMakeFiles/cv_bridge_boost.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
<== Failed to process package 'cv_bridge':
  Command '['/opt/ros/kinetic/env.sh', 'make', '-j1']' returned non-zero exit status 2

Reproduce this error by running:
==> cd /home/robot/ros_catkin_ws/build_isolated/cv_bridge && /opt/ros/kinetic/env.sh make -j1

```

I'm running Debian Jessy on a Raspberry Pi.

zhenglongyu, 2018-04-25:
i have met the same error too, have you solved it?

zhenglongyu, 2018-04-25:
I have encountered the same question.Have you fixed it?

k-okada, 2018-06-04:
does everyone using Raspberry Pi? I do not have one and tested on `osrf/ros:kinetic-desktop-jessie` docker image  but could not reproduce this. Can someone explain how to reproduce on my ubuntu/amd64 machine?

fraukej, 2018-10-02:
I had the same problem using an ubuntu 14.04 machine.

I noticed in module.hpp:
https://github.com/ros-perception/vision_opencv/blob/a3736d24554166142981fa2bcf3d7d3c6e3ce366/cv_bridge/src/module.hpp#L36-L46

the compiler tries to compile the non-python3-verision (the error occurs in line 44).
As for me, I want to compile using python3 (this might be standard for Debian Jessie?). But I don't know how to set the compiler flag `PYTHON3`.
Setting `-DPYTHON3=On` in catkin config didn't work.

What helped for me is just to delete the part that's not relevant for python3 and keep only the python3 part (without the `#if`'s and `#else`'s):

```
static int do_numpy_import( )
{
    import_array( );
}
```

I get a Warning: 
```
In file included from /home/frauke/catkin_ws/src/pallet_boxes_nn/third_party/vision_opencv/cv_bridge/src/module_opencv3.cpp:3:0:
/home/frauke/catkin_ws/src/pallet_boxes_nn/third_party/vision_opencv/cv_bridge/src/module.hpp: In function ‘int do_numpy_import()’:
/home/frauke/catkin_ws/src/pallet_boxes_nn/third_party/vision_opencv/cv_bridge/src/module.hpp:38:5: warning: converting to non-pointer type ‘int’ from NULL [-Wconversion-null]
     import_array( );
     ^
In file included from /home/frauke/catkin_ws/src/pallet_boxes_nn/third_party/vision_opencv/cv_bridge/src/module.cpp:35:0:
/home/frauke/catkin_ws/src/pallet_boxes_nn/third_party/vision_opencv/cv_bridge/src/module.hpp: In function ‘int do_numpy_import()’:
/home/frauke/catkin_ws/src/pallet_boxes_nn/third_party/vision_opencv/cv_bridge/src/module.hpp:38:5: warning: converting to non-pointer type ‘int’ from NULL [-Wconversion-null]
     import_array( );
```
 but that's fine for me.

ijnek, 2022-09-12:
I recently became the maintainer of this repo, and I'm sorting through issues, closing stale ones. As this issue has not been active for a while (and I suspect it has been fixed), I'm going to close it due to inactivity. Please feel free to comment if it is still relevant and you would like to get it re-opened.

closed 2022-09-12 by ijnek