ros-perception/vision_opencv #207: fail to use function in cv_bridge

wyixiang, 2018-04-04:
When I use some function in cv_bridge, I receive an importerror:

Traceback (most recent call last):
File "/home/yixiangw/gym-gazebo/gym_gazebo/envs/turtlebot/gazebo_circuit2c_turtlebot_camera_nn.py", line 217, in _reset
cv_image = CvBridge().imgmsg_to_cv2(image_data, "bgr8")
File "/home/yixiangw/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/vision_opencv/cv_bridge/python/cv_bridge/core.py", line 163, in imgmsg_to_cv2
dtype, n_channels = self.encoding_to_dtype_with_channels(img_msg.encoding)
File "/home/yixiangw/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/vision_opencv/cv_bridge/python/cv_bridge/core.py", line 99, in encoding_to_dtype_with_channels
return self.cvtype2_to_dtype_with_channels(self.encoding_to_cvtype2(encoding))
File "/home/yixiangw/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/vision_opencv/cv_bridge/python/cv_bridge/core.py", line 91, in encoding_to_cvtype2
from cv_bridge.boost.cv_bridge_boost import getCvType
ImportError: dynamic module does not define module export function (PyInit_cv_bridge_boost)

so how can I solve it ?

walid19, 2018-08-31:
i am having the same problem did you find the solution?

wyixiang, 2018-08-31:
@walid19 In my try, cv_bridge is available for python2 and cannot be used in py3. You can try running in python2.

Akashbaskaran, 2019-05-14:
Was anyone able to use python3 and solve this issue?

ioneliabuzatu, 2019-07-30:
> Was anyone able to use python3 and solve this issue?

Any news about it? Same issue. I use melodic

ioneliabuzatu, 2019-07-30:
Found a way around. This helped http://www.programmersought.com/article/4908420840/

qgallouedec, 2021-03-04:
You could use `numpy` directly, instead of `cv_bridge`

```python
import numpy as np
im = np.frombuffer(image_data.data, dtype=np.uint8).reshape(image_data.height, image_data.width, -1)
```

felixvd, 2021-03-08:
↑↑↑ A real MVP, just like in the legends

If you're missing the `desired_encoding` parameter, use [cv2.cvtColor](https://docs.opencv.org/master/d8/d01/group__imgproc__color__conversions.html#ga397ae87e1288a81d2363b61574eb8cab) with [these codes](https://docs.opencv.org/master/de/d25/imgproc_color_conversions.html).
```
import cv2
im_gray = cv2.cvtColor(im, cv2.COLOR_RGB2GRAY)
```

nerdneilsfield, 2021-05-31:
Is there anyway to implement the same thing as `CvBridge().cv2_to_imgmsg` rather than using `cv_bridge`?

andrewjong, 2021-07-26:
A different solution worked for me. I'm also trying to use Python 3 with CVBridge. I'm using conda as my environment manager. For me the solution was to configure catkin like so:
```
catkin config \
-DPYTHON_EXECUTABLE=/home/andrew/.miniconda3/envs/myenv/bin/python \
-DPYTHON_INCLUDE_DIR=/home/andrew/.miniconda3/envs/myenv/include/python3.7m/ \
-DPYTHON_LIBRARY=/home/andrew/.miniconda3/envs/myenv/lib/libpython3.7m.so
```
After adding these CMake Args, I no longer get the error.

kilichzf, 2021-09-20:
I was having the same error message but not with this repository.
I am writing this because it may help somebody.
In my case I forgot to source cv2_bridge package for python 3 and trying to use python 2 one.

tamerlan-b, 2021-09-30:
Beautiful solution is to use [ros_numpy](https://github.com/eric-wieser/ros_numpy). It is similar to the function of cv_bridge, but without the dependency on cv2:
```python
np_img = ros_numpy.numpify(ros_img_msg)
```

arthurBricq, 2022-06-20:
For those trying to do this in the other direction: **ndarray (or cv2) --> ROS**, without using cv_bridge, here is a snippet of code that does the job using the methods mentionned just above.

Assuming that `img` is a numpy array containing your image:  

```python
ros_image = Image(encoding="mono8")
# Create the header
ros_image.header.stamp = rospy.Time.from_sec(t)
ros_image.header.frame_id = id
# Fill the image data 
ros_image.height, ros_image.width = img.shape
ros_image.data = img.ravel().tobytes() # or .tostring()
ros_image.step=ros_image.width
```

DanielTakeshi, 2023-01-23:
@qgallouedec where are you getting `image_data` in your example code?