ros-perception/vision_opencv #208: Release into Melodic

clalancette, 2018-04-04:
It looks like all of the dependencies for vision_opencv are available in Melodic[1], so it would be great to release it.  Thanks in advance!

[1] There is still the open question of which version of opencv we use in Melodic, ala https://github.com/ros-infrastructure/rep/issues/167, but we can get vision_opencv built against the system packages while we figure that out.

clalancette, 2018-04-05:
Actually, I misunderstood how this would work, and we do need to make the decision about opencv3 before we move forward with this.  I've opened up https://github.com/ros-infrastructure/rep/pull/168 to at least get feedback.

130s, 2018-05-02:
I see https://github.com/ros/rosdistro/pull/17674, so this case resolved?

clalancette, 2018-05-02:
> I see ros/rosdistro#17674, so this case resolved?

I believe so, yes.  I'm going to close this out, we can do any follow-ups on different issues.  Thanks!

closed 2018-05-02 by clalancette