ros-perception/vision_opencv #213: buffer type issue of cv_bridge in ROS2 branch

jzp1025, 2018-04-17:
@gaoethan Hi , Thanks for your upstream in ros2 branch , i have succeed in building and import cv_bridge in my own python tests . 
But during the tests , i found that in the function : "imgmsg_to_cv2"  of the file /vision_opencv/cv_bridge/python/cv_bridge/core.py , the code is like this : 

`im = np.ndarray(shape=(img_msg.height, img_msg.width, n_channels),
                           dtype=dtype, buffer=img_msg.data)`

and it will report an error since the img_msg.data is of type "list" and the buffer needs type "Bytes"

i change the img_msg.data to np.array(img_msg.data) and i succeed in using the bridge .

So i am wondering is there a better way to solve this bug ? Or have I missed some points in the code ?

Looking forward to your opinion , thanks !

Besides , i found the opencv_tests in the ros2 branch is still using catkin , but in your [another](https://github.com/gaoethan/vision_opencv/tree/ros2_cv_bridge) gits i found an ament-builded code , so i think you might missed merge the right code in opencv_tests .

gaoethan, 2018-04-17:
@jzp1025 thanks for the issue, generally speaking, it's the requirement of the definition of `np.ndarray()` to its parameter of `buffer`, you may check its python manual for reference, so it's not a bug by converting your `img_msg.data` which is not in the format of bytes via `np.array(img_msg.data)`. so what we may do better is to insert a double checking to the `img_msg.data` **before** it's given to `buffer`.

For the `opencv_tests`, I'll migrated it here when all the cv_bridge stuff is ready. stay tuned !

jzp1025, 2018-04-17:
@gaoethan thanks for the reply . 
So have you tried to transport image using the cv_bridge in python ? i think the core.py in the gits doesn't work because when cv2 transformed to img_msg and delivered to another nodes , the img_msg.data which is subscribed will become type "list" , as i mentioned before . so only after i change the core.py code like above can i use the cv_bridge .
so i'm wondering whether it's necessary to change the core.py , or did you change something else in the code ? thanks !

gaoethan, 2018-04-17:
> i think the core.py in the gits doesn't work because when cv2 transformed to img_msg and delivered to another nodes , the img_msg.data which is subscribed will become type "list" , as i mentioned before 

Sorry, I'm a little confused regarding this description, let's make it clear, do you mean the following:
A ros2 node publishes the `img_msg.data` which is type of list, then this `img_msg.data` is given to python API `imgmsg_to_cv2()`, but `imgmsg_to_cv2()` can't proceed your `img_msg.data` before you covert `img_msg.data` with np.array() to type bytes?

jzp1025, 2018-04-18:
> A ros2 node publishes the img_msg.data which is type of list, then this img_msg.data is given to python API imgmsg_to_cv2(), but imgmsg_to_cv2() can't proceed your img_msg.data before you covert img_msg.data with np.array() to type bytes?

yes .
because in ROS , type uint8 will be convert to string when it is given to python so there is no error (actually other type will be convert to list too in ROS)
but in ROS2 , all type will be convert to list including uint8 , so we need to convert `img_msg.data` with np.array() before we use `im = np.ndarray(shape=(img_msg.height, img_msg.width, n_channels), dtype=dtype, buffer=img_msg.data)` , cause buffer needs type bytes
so in conclusion , the core.py now is not work in passing image between two nodes in python . my team was wondering whether we should do this conversion from list to byte ?
thanks !

gaoethan, 2018-04-18:
> so in conclusion , the core.py now does not work in passing image between two nodes in python.

The image passing btw two nodes thru topic and it's independent of the conversion to OpenCV `Mat`, so how does it affect the message exchange with the output of `imgmsg_to_cv2()` ? thanks !

BTW, I'd like to check whether your data can pass the checking of `@data.setter` of `img_msg.data` which is generated from rosidl_generator_py/resource/_msg.py.em in ros2:
```
    @data.setter
    def data(self, value):
        from collections import Sequence
        from collections import Set
        from collections import UserList
        from collections import UserString
        assert \
            ((isinstance(value, Sequence) or
              isinstance(value, Set) or
              isinstance(value, UserList)) and
             not isinstance(value, str) and
             not isinstance(value, UserString) and
             all(isinstance(v, int) for v in value) and
             all(val >= 0 and val < 256 for val in value)), \
            "The 'data' field must be a set or sequence and each value of type 'int' and each unsigned integer in [0, 255]"
        self._data = value

```

jzp1025, 2018-04-18:
> The image passing btw two nodes thru topic and it's independent of the conversion to OpenCV Mat, so how does it affect the message exchange with the output of imgmsg_to_cv2() ? 
It's not the exchange btw two nodes that have error . the data can be transport to the destination node
It's after the exchange btw twp nodes , the core.py try to convert the received img_msg to cv2 in func `imgmsg_to_cv2()` that occur errors . 
so as i said above , the problem is ：
        in ROS , the `uint8` type msg convert to another node will be in type string in python , but in ROS2 , it will be turned to list . It's a characteristic of ROS2 
         so the core.py work in ROS , but actually need a np.array to make it work in ROS2 , due to the `buffer = img_msg.data`

And after i add a np.array to the `imgmsg_to_cv2()` , i have succeed transport images btw two nodes , and use cv2.imshow to display .
Looking forward to your opinion

@gaoethan thanks

gaoethan, 2018-04-18:
Ok. Just as the `@data.setter` definition of ros2 `img_msg.data` aforementioned, you can see it has asserted the data can be instance of `sequence,set, userlist` but not `str and userstring`. so it makes sense that the type string (in ros previously) converted doesn't work now in ros2 ! :)

gaoethan, 2018-04-19:
thanks for your details, and I think maybe something tricky for [img_msg.data setting](https://github.com/ros-perception/vision_opencv/blob/ros2/cv_bridge/python/cv_bridge/core.py#L263-L264) in API `cv2_to_imgmsg()` in ros2, and you can add a checking and conversion in your callback `detect_and_draw()` before given to `imgmsg_to_cv2()`  to workaround temporarily.

jzp1025, 2018-04-19:
> and you can add a checking and conversion in your callback detect_and_draw() before given to imgmsg_to_cv2()

So in conclusion your opinion is ：
        i still need to convert  the img_msg.data from list to bytes ， not in core.py ,  but in callback funtion 
        i think one way is to copy all data in img_msg except img_msg.data to another variable `tmp_data` and make `tmp_data.data = np.array(img_msg.data）`

i want to configure  :
         will  the bug be fixed some day or we just deal the img_msg like i mentioned above in callback function , cause the core.py can't work with the features in ROS2 properly

thanks! @gaoethan

pantor, 2018-07-19:
Hi there, is there any news on this issue in the meanwhile? I also need to use `np.array(img_msg.data)` for receiving images in ros2.

gaoethan, 2018-07-26:
This issue have already been enhanced in PR #235 and close it.

closed 2018-07-26 by gaoethan

pankhurivanjani, 2019-08-09:
Hello @gaoethan!
 I am using ros2-Dashing and getting the same error.

```    File "/opt/ros/dashing/lib/python3.6/site-packages/cv_bridge/core.py", line 184, in imgmsg_to_cv2
    dtype=dtype, buffer=img_buf)
TypeError: buffer is too small for requested array ```
 core.py has also been fixed with the solution given in the above discussion as in [dashing-core.py](https://github.com/ros-perception/vision_opencv/blob/dashing/cv_bridge/python/cv_bridge/core.py#L173)

I checked the type of buffer and it was " class 'array.array' " What can we do ahead in it?

Thanks,

ambareeshsrja16, 2019-08-17:
Hi,
I made the same change:
buf = np.asarray(img_msg.data, dtype = dtype)
and assigned it to buffer. However, I'm getting the following error:

ValueError: invalid literal for int() with base 10: b'\n\x0c\x00\r\x0f\x03\x0e\x0e\x05\r\x0c\x04\x0e\n\x06\x0b\x07\x04\x0b\n\x01\t\x08\x00\x03\x0e\x03\x05\x10\x06\x00\r\x04\x00\x0c\x03\n\x0c\x02\n\x0c\x02\x14\x0f\x03\x10\x0c\x00\x0b\x10\x01\x0c\x11\x02

Any suggestions?

Thank you!

pankhurivanjani, 2019-08-17:
Hello ! In my case I was sending the compressed file via publisher and expecting a normal image. This was creating error of buffer size. 

Current core.py in Ros2 dashing is perfectly fine, no changes are required.

ambareeshsrja16, 2019-08-18:
Hi,

Unfortunately, I have to work with ROS at the moment. I am directly playing a video stream from a bag (no compression involved) I am also noticing that it doesn't happen for all images, just a few and far between.

PR #235 (2018-07-26): Fix buffer type error in ROS2
This fixes #213 for me.