ros-perception/vision_opencv #215: Cannot locate Boost-Python

Vibrane, 2018-05-17:
[image]

I am having the above problem while i have the following:

[image]

Totally confused what to do.

jimu-dongyu, 2018-08-01:
@Vibrane Have you solved your problem? I have the same problem as you.

meyerj, 2018-08-02:
From the screenshots I assume you are compiling with Boost 1.67 from HomeBrew. In this version the `boost_python` library has been renamed to `boost_python27`:
https://www.boost.org/doc/libs/1_67_0/libs/python/doc/html/rn.html

The cmake module `FindBoost.cmake` cannot cope with that if you do not add the version suffix to the library name in cmake:
```cmake
# fails:
find_package(Boost REQUIRED python)
# works:
find_package(Boost REQUIRED python27)
```
(tested with CMake 3.12 installed from HomeBrew).

CMake 3.11 added the following to the [FindBoost](https://cmake.org/cmake/help/v3.12/module/FindBoost.html) documentation:
> Note that Boost Python components require a Python version suffix (Boost 1.67 and later), e.g. python36 or python27 for the versions built against Python 3.6 and 2.7, respectively. This also applies to additional components using Python including mpi_python and numpy. Earlier Boost releases may use distribution-specific suffixes such as 2, 3 or 2.7. These may also be used as suffixes, but note that they are not portable.

I am currently preparing a pull request. The same issue affects the [camera_calibration_parsers](https://github.com/ros-perception/image_common/blob/hydro-devel/camera_calibration_parsers/CMakeLists.txt) package in [image_common](https://github.com/ros-perception/image_common).

ijnek, 2022-09-12:
I recently became the maintainer of this repo, and I'm sorting through issues, closing stale ones. As this issue has not been active for a while, I'm going to close it due to inactivity. Please feel free to comment if it is still relevant and you would like to get it re-opened.

closed 2022-09-12 by ijnek

apusky, 2023-11-01:
I am trying to build the library and I get the same error:
`--- stderr: cv_bridge                                                                                                                                                                                     
CMake Error at /usr/lib/aarch64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake:117 (find_package):
  Could not find a package configuration file provided by "boost_python"
  (requested version 1.71.0) with any of the following names:

    boost_pythonConfig.cmake
    boost_python-config.cmake

  Add the installation prefix of "boost_python" to CMAKE_PREFIX_PATH or set
  "boost_python_DIR" to a directory containing one of the above files.  If
  "boost_python" provides a separate development package or SDK, be sure it
  has been installed.
Call Stack (most recent call first):
  /usr/lib/aarch64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake:182 (boost_find_component)
  /usr/local/lib/python3.8/dist-packages/cmake/data/share/cmake-3.22/Modules/FindBoost.cmake:594 (find_package)
  CMakeLists.txt:34 (find_package)

`