ros-perception/vision_opencv #224: opencv3 build issue

doll6777, 2018-06-15:
****

I encoountered error. I installed opencv3 3.4.1 and checked It is not muti-installed with opencv 2.4.1
I think when i am trying to build, it reads opencv3 file but try to build with opencv2. 
can i know the reason?

/home/sml/ros2_ws/src/vision_opencv/cv_bridge/src/module_opencv3.cpp: In member function ‘void NumpyAllocator::allocate(int, const int*, int, int*&, uchar*&, uchar*&, size_t*)’:
/home/sml/ros2_ws/src/vision_opencv/cv_bridge/src/module_opencv3.cpp:121:34: error: expected ‘(’ before ‘)’ token
   const int f = (static_cast<int>)(sizeof( size_t ) / 8);
                                  ^
/home/sml/ros2_ws/src/vision_opencv/cv_bridge/src/module_opencv3.cpp:121:34: error: expected primary-expression before ‘)’ token
/home/sml/ros2_ws/src/vision_opencv/cv_bridge/src/module_opencv3.cpp:121:57: error: expected ‘)’ before ‘;’ token
   const int f = (static_cast<int>)(sizeof( size_t ) / 8);
                                                         ^
/home/sml/ros2_ws/src/vision_opencv/cv_bridge/src/module_opencv3.cpp: At global scope:
/home/sml/ros2_ws/src/vision_opencv/cv_bridge/src/module_opencv3.cpp:150:57: warning: unused parameter ‘datastart’ [-Wunused-parameter]
 void NumpyAllocator::deallocate(int * refcount, uchar * datastart, uchar * data)
                                                         ^
/home/sml/ros2_ws/src/vision_opencv/cv_bridge/src/module_opencv3.cpp:150:76: warning: unused parameter ‘data’ [-Wunused-parameter]
 void NumpyAllocator::deallocate(int * refcount, uchar * datastart, uchar * data)
                                                                            ^
/home/sml/ros2_ws/src/vision_opencv/cv_bridge/src/module_opencv3.cpp:161:16: error: cannot declare variable ‘g_numpyAllocator’ to be of abstract type ‘NumpyAllocator’
 NumpyAllocator g_numpyAllocator;
                ^
/home/sml/ros2_ws/src/vision_opencv/cv_bridge/src/module_opencv3.cpp:102:7: note:   because the following virtual functions are pure within ‘NumpyAllocator’:
 class NumpyAllocator : public cv::MatAllocator
       ^
In file included from /usr/local/include/opencv2/core.hpp:59:0,
                 from /usr/local/include/opencv2/core/core.hpp:48,
                 from /home/sml/ros2_ws/src/vision_opencv/cv_bridge/include/cv_bridge/cv_bridge.h:43,
                 from /home/sml/ros2_ws/src/vision_opencv/cv_bridge/src/module.hpp:21,
                 from /home/sml/ros2_ws/src/vision_opencv/cv_bridge/src/module_opencv3.cpp:37:
/usr/local/include/opencv2/core/mat.hpp:452:23: note: 	virtual cv::UMatData* cv::MatAllocator::allocate(int, const int*, int, void*, size_t*, int, cv::UMatUsageFlags) const
     virtual UMatData* allocate(int dims, const int* sizes, int type,
                       ^
/usr/local/include/opencv2/core/mat.hpp:454:18: note: 	virtual bool cv::MatAllocator::allocate(cv::UMatData*, int, cv::UMatUsageFlags) const
     virtual bool allocate(UMatData* data, int accessflags, UMatUsageFlags usageFlags) co
                  ^
/usr/local/include/opencv2/core/mat.hpp:455:18: note: 	virtual void cv::MatAllocator::deallocate(cv::UMatData*) const
     virtual void deallocate(UMatData* data) const = 0;
                  ^
/home/sml/ros2_ws/src/vision_opencv/cv_bridge/src/module_opencv3.cpp: In function ‘int convert_to_CvMat2(const PyObject*, cv::Mat&)’:
/home/sml/ros2_ws/src/vision_opencv/cv_bridge/src/module_opencv3.cpp:181:65: error: reinterpret_cast from type ‘const PyObject* {aka const _object*}’ to type ‘PyArrayObject* {aka tagPyArrayObject*}’ casts away qualifiers
   int typenum = PyArray_TYPE(reinterpret_cast<PyArrayObject *>(o));
                                                                 ^
/home/sml/ros2_ws/src/vision_opencv/cv_bridge/src/module_opencv3.cpp:193:63: error: reinterpret_cast from type ‘const PyObject* {aka const _object*}’ to type ‘PyArrayObject* {aka tagPyArrayObject*}’ casts away qualifiers
   int ndims = PyArray_NDIM(reinterpret_cast<PyArrayObject *>(o));
                                                               ^
/home/sml/ros2_ws/src/vision_opencv/cv_bridge/src/module_opencv3.cpp:201:77: error: reinterpret_cast from type ‘const PyObject* {aka const _object*}’ to type ‘PyArrayObject* {aka tagPyArrayObject*}’ casts away qualifiers
   const npy_intp * _sizes = PyArray_DIMS(reinterpret_cast<PyArrayObject *>(o));
                                                                             ^
/home/sml/ros2_ws/src/vision_opencv/cv_bridge/src/module_opencv3.cpp:202:82: error: reinterpret_cast from type ‘const PyObject* {aka const _object*}’ to type ‘PyArrayObject* {aka tagPyArrayObject*}’ casts away qualifiers
   const npy_intp * _strides = PyArray_STRIDES(reinterpret_cast<PyArrayObject *>(o));
                                                                                  ^
/home/sml/ros2_ws/src/vision_opencv/cv_bridge/src/module_opencv3.cpp:206:32: error: expected ‘(’ before ‘)’ token
     size[i] = (static_cast<int>)_sizes[i];
                                ^
/home/sml/ros2_ws/src/vision_opencv/cv_bridge/src/module_opencv3.cpp:206:32: error: expected primary-expression before ‘)’ token
/home/sml/ros2_ws/src/vision_opencv/cv_bridge/src/module_opencv3.cpp:206:33: error: expected ‘)’ before ‘_sizes’
     size[i] = (static_cast<int>)_sizes[i];
                                 ^
/home/sml/ros2_ws/src/vision_opencv/cv_bridge/src/module_opencv3.cpp:232:82: error: reinterpret_cast from type ‘const PyObject* {aka const _object*}’ to type ‘PyArrayObject* {aka tagPyArrayObject*}’ casts away qualifiers
  = cv::Mat(ndims, size, type, PyArray_DATA(reinterpret_cast<PyArrayObject *>(o)), step);
                                                                               ^
/home/sml/ros2_ws/src/vision_opencv/cv_bridge/src/module_opencv3.cpp:235:7: error: ‘class cv::Mat’ has no member named ‘refcount’
     m.refcount = refcountFromPyObject(o);
       ^
/home/sml/ros2_ws/src/vision_opencv/cv_bridge/src/module_opencv3.cpp:201:20: warning: unused variable ‘_sizes’ [-Wunused-variable]
   const npy_intp * _sizes = PyArray_DIMS(reinterpret_cast<PyArrayObject *>(o));
                    ^
/home/sml/ros2_ws/src/vision_opencv/cv_bridge/src/module_opencv3.cpp: In function ‘PyObject* pyopencv_from(const cv::Mat&)’:
/home/sml/ros2_ws/src/vision_opencv/cv_bridge/src/module_opencv3.cpp:256:11: error: ‘class cv::Mat’ has no member named ‘refcount’
   if (!p->refcount || p->allocator != &g_numpyAllocator) {
           ^
/home/sml/ros2_ws/src/vision_opencv/cv_bridge/src/module_opencv3.cpp:262:34: error: ‘class cv::Mat’ has no member named ‘refcount’
   return pyObjectFromRefcount(p->refcount);
                                  ^
In file included from /home/sml/ros2_ws/src/vision_opencv/cv_bridge/src/module_opencv3.cpp:37:0:
/home/sml/ros2_ws/src/vision_opencv/cv_bridge/src/module.hpp: In function ‘int do_numpy_import()’:
/home/sml/ros2_ws/src/vision_opencv/cv_bridge/src/module.hpp:38:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^
make[2]: *** [src/CMakeFiles/cv_bridge_boost.dir/module_opencv3.cpp.o] Error 1
src/CMakeFiles/cv_bridge_boost.dir/build.make:86: recipe for target 'src/CMakeFiles/cv_bridge_boost.dir/module_opencv3.cpp.o' failed
make[1]: *** [src/CMakeFiles/cv_bridge_boost.dir/all] Error 2
CMakeFiles/Makefile2:1070: recipe for target 'src/CMakeFiles/cv_bridge_boost.dir/all' failed
make: *** [all] Error 2
<== Command '. /home/sml/ros2_ws/build/cv_bridge/cmake__build.sh && /usr/bin/make -j8 -l8' failed in '/home/sml/ros2_ws/build/cv_bridge' with exit code '2'
<== Command '. /home/sml/ros2_ws/build/cv_bridge/cmake__build.sh && /usr/bin/make -j8 -l8' failed in '/home/sml/ros2_ws/build/cv_bridge' with exit code '2'

gaoethan, 2018-07-06:
Sorry, I don't see it directs to the opencv2 from the log ?  I advised that you only install opencv3 with a clear   environment to double check,  thanks !

ijnek, 2022-09-12:
I recently became the maintainer of this repo, and I'm sorting through issues, closing stale ones. As this issue has not been active for a while, I'm going to close it due to inactivity. Please feel free to comment if it is still relevant and you would like to get it re-opened.

closed 2022-09-12 by ijnek