ros-perception/vision_opencv #236: cv_bridge boost error

pnambiar, 2018-07-21:
I am trying to create a ROS wrapper for caffe2 and I get the following error:
[ERROR] [1532132328.463265]: bad callback: <bound method image_converter.callback of <__main__.image_converter instance at 0x7f01bdc8e7a0>>
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/topics.py", line 750, in _invoke_callback
    cb(msg)
  File "ros_mercnn.py", line 72, in callback
    cv_image = self.bridge.imgmsg_to_cv2(data, 'bgr8')
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/cv_bridge/core.py", line 182, in imgmsg_to_cv2
    res = cvtColor2(im, img_msg.encoding, desired_encoding)
ArgumentError: Python argument types in
    cv_bridge.boost.cv_bridge_boost.cvtColor2(numpy.ndarray, str, unicode)
did not match C++ signature:
    cvtColor2(boost::python::api::object, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)

Any help will be highly appreciated

#!/usr/bin/env python

"""Perform inference on a single image or all images with a certain extension
(e.g., .jpg) in a folder.
"""

from __future__ import absolute_import
from __future__ import division
from __future__ import print_function
from __future__ import unicode_literals

from collections import defaultdict
import argparse
import cv2  # NOQA (Must import before importing caffe2 due to bug in cv2)
import glob
import logging
import os
import sys
import time

from caffe2.python import workspace

import roslib
#roslib.load_manifest('my_package')
import sys
import rospy
#import cv2
from std_msgs.msg import String
from sensor_msgs.msg import Image
from cv_bridge import CvBridge, CvBridgeError
import numpy as np
class image_converter:	

	def __init__(self):

		self.bridge = CvBridge()
		self.image_sub = rospy.Subscriber("/usb_cam/image_raw",Image,self.callback)
		cv2.waitKey(1000000)
	def callback(self, data):
		currentframe = 0
		try:
			while True:
 
				cv_image = self.bridge.imgmsg_to_cv2(data, 'bgr8')

				
		except CvBridgeError as e:	
			print(e)
def main(args):
	ic = image_converter()
	rospy.init_node('image_converter', anonymous=True)
	try:
		cv2.waitKey(1000)
		rospy.spin()
	except KeyboardInterrupt:
		print('Shutting Down')
	cv2.destroyAllWindows()

if __name__ == '__main__':

        workspace.GlobalInit(['caffe2', '--caffe2_log_level=0'])
  #      setup_logging(__name__)
	main(sys.argv)

nslay, 2018-08-10:
Just change File "/opt/ros/kinetic/lib/python2.7/dist-packages/cv_bridge/core.py", line 182
res = cvtColor2(im, img_msg.encoding, desired_encoding)
to
res = cvtColor2(im, img_msg.encoding, str(desired_encoding))

Did the trick for me! Very lame... I hate Python.

asimonov, 2021-04-29:
I am seeing this issue on Noetic on ARM64 (Jetson NX)

I run ubuntu20 in docker and see this (with default cv_bridge)

```
# source /opt/ros/noetic/setup.bash 
# roscd cv_bridge
/opt/ros/noetic/share/cv_bridge# python
Python 3.8.5 (default, Jan 27 2021, 15:41:15) 
[GCC 9.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import cv_bridge
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/opt/ros/noetic/lib/python3/dist-packages/cv_bridge/__init__.py", line 6, in <module>
    from cv_bridge.boost.cv_bridge_boost import cvtColorForDisplay, getCvType
SystemError: initialization of cv_bridge_boost raised unreported exception
>>> 
```

and if I take current master and build it in my workspace I see the same issue:

```
root@uc1-nx-1:~/git/catkin_ws# source devel/setup.bash 
root@uc1-nx-1:~/git/catkin_ws# roscd cv_bridge
root@uc1-nx-1:~/git/catkin_ws/src/vision_opencv/cv_bridge# python
Python 3.8.5 (default, Jan 27 2021, 15:41:15) 
[GCC 9.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> python
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'python' is not defined
>>> import cv_bridge
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/tii/git/catkin_ws/devel/lib/python3/dist-packages/cv_bridge/__init__.py", line 34, in <module>
    exec(__fh.read())
  File "<string>", line 6, in <module>
SystemError: initialization of cv_bridge_boost raised unreported exception
>>> 
```

Any ideas how to debug/fix?

ijnek, 2022-09-12:
I recently became the maintainer of this repo, and I'm sorting through issues, closing stale ones. As this issue has not been active for a while, I'm going to close it due to inactivity. Please feel free to comment if it is still relevant and you would like to get it re-opened.

For reference, [ROS Answers](https://answers.ros.org/questions/) would be the place to post such questions.

closed 2022-09-12 by ijnek