ros-perception/vision_opencv #247: cv_bridge_boost library missing on Windows

edualvarado, 2018-09-05:
I want to use the package `cv_bridge` in a Python script running on Windows 10 x64. I do not have Kinectic installed. I just need to import some packages to my program, such as `rosbag`, among others (In summary, my script takes video-data from a .bag file, and shows it on a GUI using OpenCV, all running on Windows).

I already installed the package in my `C:\Python27\Lib\site-packages` by running the `setup.py` file. When I run the Python script, the following error comes out:

> from cv_bridge.boost.cv_bridge_boost import getCvType
ImportError: No module named boost.cv_bridge_boost

I realized that the installation of this package on Windows didn't create the folder `boost `in the installation path, while in Ubuntu it exits. The folder contains the requested library, `cv_brigde_boost.so.` But as you can notice, this is a Linux format, so when I copy it to my package path on Windows, the error is still coming out, because it doesn't recognize it.

Is it normal that the library` cv_brigde_boost` is not being created for Windows when I install the package? Or is there maybe any sort of missing dependency that I need? I am really stuck here, and I don't find much info about it. Thank you very much in advance.

kaalessi, 2020-06-04:
@edualvarado Did you find a solution to this? I am doing a similar thing as you and have hit the same bump in the road.

AndreasGronlund, 2021-02-19:
@edualvarado @kaalessi Any solution? I have the same issue.

MikeHallettUK, 2021-02-25:
Me too ... on Python 3.8 and NOETIC

kaalessi, 2021-02-26:
I still have not found a solution. The project I was working on that this could have been used was abandoned.

MikeHallettUK, 2021-02-28:
I just raiswed this bug at: [https://github.com/ms-iot/ROSOnWindows/issues/317](https://github.com/ms-iot/ROSOnWindows/issues/317)