ros-perception/vision_opencv #251: cv_bridge crash with ROS2 Python APIs

alsora, 2018-10-28:
Hi,

I'm using cv_bridge in some ROS2 Bouncy projects.

If I use the C++ APIs, everything works fine.
On the other hand with the Python ones my program crashes with no error message.

```
import numpy as np
import cv2
import rclpy
from cv_bridge import CvBridge

def main(args=None):

    #rclpy.init(args=args)

    br = CvBridge()

    dtype, n_channels = br.encoding_to_dtype_with_channels('8UC3')

    im = np.ndarray(shape=(480, 640, n_channels), dtype=dtype)

    msg = br.cv2_to_imgmsg(im)
    im2 = br.imgmsg_to_cv2(msg)

if __name__ == '__main__':
    main()
```

If I try to run the following code, it crashes during `br.encoding_to_dtype_with_channels('8UC3')`.

Moreover note that I'm not able to call `rclpy.init(args=args)` if I'm importing `CvBridge`.

The system is generated through [this](https://github.com/alsora/ros2-tensorflow/blob/master/docker/Dockerfile) Dockerfile

As a recap, I am using 
- Ubuntu 16.04
- ROS2 Bouncy 
- Python 3.5.2
- OpenCV Python 3.4.3

gaoethan, 2018-10-31:
It works to call `rclpy.init()` after importing `CvBridge` before, maybe you can refer to an example https://github.com/ros-perception/vision_opencv/blob/ros2/opencv_tests/nodes/source.py. thanks !

gaoethan, 2018-11-01:
It's a little strange, because it works before and not sure whether it relates to the other latest changes, it needs further investigation, on the other hand, actually the ROS2 version has almost removed the using of Boost and used C++ 11 or later instead.

alsora, 2018-11-01:
I tried to build and run the tests checking out old commits. 
Tags `2.0.5` and `2.0.4` are not working either.

I also tried updating boost to 1.6.3 but the result does not change

8489faustn, 2019-05-05:
Were you ever able to figure out the fix? Currently having the same exact issue when trying to run camera calibration for a kinect camera. Keeps giving me the segmentation error when trying to run `self.br.encoding_to_dtype_with_channels(msg.encoding)[0]`

ixtiyoruz, 2020-04-24:
I also have the same issue. Is this bug resolved?