ros-perception/vision_opencv #255: Unable to restore Mono 16 format image in cv_bridge of ros 2

onishu, 2018-11-29:
I use realsense D415 to get depth image (mono16).
(topic name: /camera/aligned_depth_to_color/image_raw)
In ros2 (rclpy), I subscribe to the message (sensor_msgs::Image) but the image is 8 bits.

I check "imgmsg_to_cv2" function (/python/cv_bridge/core.py).
In this case this function run "n_channels == 1".
So, "im" has only 8bit data.
Is this cause?

If you add the following code to 181 line, I get correct depth image(mono16).
Am I wrong?

``` python
if(dtype == "uint16"):
    im = np.ndarray(shape=(img_msg.height, img_msg.width, 2),
                    dtype=dtype, buffer=img_buf)
    im = (im[:,:,1] << 8) + im[:,:,0]
```

gaoethan, 2018-11-29:
Pls try to file a PR if you have the right solution to fix the issue, thanks !

onishu, 2018-11-29:
I think this solution is interim measures. Because this code judged by data type.
The right solution should be modified based on design policy.(ex: judge by "img_msg.encoding" or "n_channels" )
I need permanent measure.

CaffeMrDe, 2019-07-18:
system:  ubuntu16.04
ros :        kinect
camera:  kinect camera

I am coding own camera Api. Met similar question about it.

def colorImgCB(colorData,depthData):
    #........
    filecad= folder+"JPEGImages/%s.jpg" % FileName
    filedepth= folder+"depth/%s.npy" % FileName
    FileName = FileName + 1

    #cv_img = bridge.imgmsg_to_cv2(data, "bgr8")
    **depth_image = bridge.imgmsg_to_cv2(depthData, 'mon16')**
    color_image = bridge.imgmsg_to_cv2(colorData, "bgr8")
   #.........

issue export:
```
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/cv_bridge/core.py", line 184, in imgmsg_to_cv2
    raise CvBridgeError(e)
CvBridgeError: [16UC1] is not a color format. but [mono16] is. The conversion does not make sense
```

I also think it is a bug. Please repair it . Thanks

0wu, 2023-10-03:
https://github.com/ros-perception/vision_opencv/blob/0b017fe32b9ac73631bbd74a9edf75394df191c0/cv_bridge/src/cv_bridge.cpp#L219-L222

Maybe the `enc::isMono(src_encoding)` should be `!enc::isMono(src_encoding)`