ros-perception/vision_opencv #258: Mac: cv_bridge, find_package boost_python3 not found

brawner, 2018-12-14:
Build of ```cv_bridge``` failed for me on my mac (10.13.6), I installed boost-python3 with homebrew (Python 3.7)

Its CMakeLists.txt searches for 
```
find_package(Boost REQUIRED python3)
```

I found the following allowed for it to find the proper library.
```
find_package(Boost REQUIRED python37)
```

mikepurvis, 2019-03-27:
This went in in 5ae9728324dfeb8e4ef5925727f5ee894588920c, and the situation is actually even worse— see https://github.com/ros-perception/image_common/pull/85 for some discussion, but the executive summary is:

- For Python 2, you always want `find_package(Boost REQUIRED COMPONENTS python)`.
- For Python 3 before Boost 1.67 (eg Ubuntu Xenial), you want `find_package(Boost REQUIRED COMPONENTS python-py35)`.
- For Python 3 on Boost 1.67+, you want `find_package(Boost REQUIRED COMPONENTS python35)`.

You can suck in `PythonInterp` to get the version number parts and generate the string correctly, but you have to still have `PythonLibs` there or the include path won't get populated for `pyconfig.h`.

ijnek, 2022-09-12:
I recently became the maintainer of this repo, and I'm sorting through issues, closing stale ones. As this issue has not been active for a while, I'm going to close it due to inactivity. Please feel free to comment if it is still relevant and you would like to get it re-opened.

closed 2022-09-12 by ijnek