ros-perception/vision_opencv #261: cv_bridge ros2 compilation failing in Ubuntu16

nzlz, 2018-12-29:
Hi, I am not able to compile cv_bridge (ros2 branch) in Ubuntu 16.04. Ive tried many different opencv versions, installations etc.. but Im still not able to make it work. I want to avoid installing opencv from source since it takes too long.
```
colcon build --merge-install --packages-select cv_bridge
Starting >>> cv_bridge
--- stderr: cv_bridge                         
CMake Error at CMakeLists.txt:33 (find_package):
  Could not find a package configuration file provided by "OpenCV" (requested
  version 3) with any of the following names:

    OpenCVConfig.cmake
    opencv-config.cmake

  Add the installation prefix of "OpenCV" to CMAKE_PREFIX_PATH or set
  "OpenCV_DIR" to a directory containing one of the above files.  If "OpenCV"
  provides a separate development package or SDK, be sure it has been
  installed.

---
Failed   <<< cv_bridge	[ Exited with code 1 ]
```
After I install libopencv-dev

```
Starting >>> cv_bridge
--- stderr: cv_bridge                         
CMake Error at CMakeLists.txt:33 (find_package):
  Could not find a configuration file for package "OpenCV" that is compatible
  with requested version "3".

  The following configuration files were considered but not accepted:

    /usr/share/OpenCV/OpenCVConfig.cmake, version: 2.4.9.1
    /usr/share/OpenCV/OpenCVConfig.cmake, version: 2.4.9.1

---
Failed   <<< cv_bridge	[ Exited with code 1 ]
```
As I said, I tried other solutions  but none worked.

Any guidelines or comments? Thanks!

gaoethan, 2019-01-02:
From the log, it seems that the OpenCV version in your system is 2.4.x but not OpenCV 3.x

nzlz, 2019-01-02:
I have already installed 3.4.5 version, but I can't manage to use it for the compilation. I will try installing this same version from source and updating the OpenCV_DIR, but I don't really want to do this since it will make the installation process of some instructions I'm creating much longer.

```
>>> cv2.__version__
'3.4.5'
```
Ideally I would install opencv via `pip3 install opencv-python`.

nzlz, 2019-01-02:
Following [this](https://github.com/milq/milq/blob/master/scripts/bash/install-opencv.sh) installation script I managed to compile cv_bridge. But it would be great if there is a quicker solution.

gaoethan, 2019-01-03:
Generally speaking, the opencv is in the list of ros dependency configuration for ubuntu 16, so it should not require much to do make it work, I suggest that you run it within a docker with a clean Ubuntu 16 image have a check whether it results from your system environment, anyway, I'll have a check with a clean docker to install and run it later, thanks

ijnek, 2022-09-12:
I recently became the maintainer of this repo, and I'm sorting through issues, closing stale ones. As this issue is raised against a platform that is no longer supported in any of the active ros 2 distros, I'm going to close it out. Please feel free to comment if it is still relevant and you would like to get it re-opened.

closed 2022-09-12 by ijnek