ros-perception/vision_opencv #273: [Windows][melodic] add Windows build support

seanyen-msft, 2019-03-08:
This is an enhancement ticket for `vision_opencv`. I am using this ticket to track all the required work to bring up `vision_opencv` to be able to build & run on Windows/MSVC (which is verified in https://aka.ms/ros project). And hope the maintainers can take a look. Thanks!

- [ ] (nice-have) [melodic] Fix travis CI #269 
- [ ] (must-have) [Windows][melodic] remove path splash separator from 'package_dir' #267 
- [ ] (must-have) add DLL import/export macros #266
- [ ] (must-have) update CMakeLists.txt for Windows build environment #265

seanyen-msft, 2019-03-08:
@vrabaud

mjcarroll, 2020-03-28:
I have merged all corresponding PRs, so I'm going to close this.

closed 2020-03-28 by mjcarroll