ros-perception/vision_opencv #289: [ros2] cv_bridge boost dependency

jnoyola, 2019-09-06:
**Problem**
I'm trying to build the dashing branch but cv_bridge fails with the error:

> cv_bridge\src\cv_bridge.cpp(39,10): error C1083:  Cannot open include file: 'boost/endian/conversion.hpp'

I see that PR #212 intended to "remove boost dependence", removing Boost from the include_directories call, but `cv_bridge\src\cv_bridge.cpp` [still contains the include](https://github.com/ros-perception/vision_opencv/blob/b19d3d10c0c822c454047c1c4950ab22dc4a9b93/cv_bridge/src/cv_bridge.cpp#L39) (in fact the same PR updated that line).

**Fix**
If I add back the include to be `include_directories(include ${Boost_INCLUDE_DIRS})`, then it builds successfully.

**Question**
So my question is: am I doing something wrong or is it possible the ros2 branches haven't built in over a year?

I'm on Windows 10, but based on the evidence it shouldn't matter.

mjcarroll, 2019-09-06:
You are correct, it needs to be there for Windows for the current build.  I believe that the header gets found correctly on Linux because it's in the `/usr/include` or `/usr/local/include` based on the distribution.

I added a shim to rcpputils to remove the boost/endian dependency (https://github.com/ros2/rcpputils/blob/master/include/rcpputils/endian.hpp), we could potentially get that backported to dashing to completely remove boost dependency from C++, but there won't be any python until pybind11 lands.

mjcarroll, 2019-09-06:
That was an oversight, I think that the cv_bridge package is blacklisted in the ROS2 CI on Windows, so I may have missed it.

jnoyola, 2019-09-06:
I don't fully understand pybind11 work, but I believe it's working fine for me on Windows with boost::python, although I see how it can be more difficult to maintain.

So what's the best path forward here? I would love to see dashing in a buildable state, even if it still has the boost::python dependency in the interim.

mjcarroll, 2019-09-06:
Let's add `include_directories(include ${Boost_INCLUDE_DIRS})` back for now, it shouldn't have any impact.

mjcarroll, 2019-09-06:
> I don't fully understand pybind11 work, but I believe it's working fine for me on Windows with boost::python, although I see how it can be more difficult to maintain.

Generally, we're trying to avoid bringing in the Boost dependency in ROS2, if possible.  C++11 and beyond add much of the functionality that ROS was using from Boost, and it keeps the dependency tree much smaller.

PyBind11 provides generally the same functionality as Boost::Python without the dependency.

jnoyola, 2019-09-06:
If that's the case, should we reconsider #277? I don't have insight into the other changes to include OpenCV and export symbols. If that's overkill for now then we can just add the one line.

jnoyola, 2019-09-30:
Ping @mjcarroll, if you could please take a look at one of those 2 PRs

jnoyola, 2019-11-05:
Resolved with #290

closed 2019-11-05 by jnoyola