ros-perception/vision_opencv #291: error installing vision_opencv using catkin_make with OpenCV 4.1.0

SalilVishnuKapur, 2019-09-18:
Issue :
I have compiled opencv 4.1.0 with cuda support. So want to build the ros dependent packages with custom opencv. The command after initializing catkin_init_workspace is as follows. Any kind of help will be deeply appreciated
Command :
catkin_make -DCMAKE_BUILD_TYPE=Release -DOpenCV_DIR=/opt/include/opencv4/opencv2 -DPYTHON_EXECUTABLE=/usr/bin/python3

My fixes till yet :
I fixed the initial issues by editing the cmake file to have opencv version 4.1.0 and also /home/spiri/ros_opencv_dependables/src/orb_slam_2_ros/orb_slam2/include/ORBextractor.h to have #include <opencv2/opencv.hpp> instead of #include <opencv/cv.h>

My system config :
Ubuntu 18.04
Opencv 4.1.0
gcc 7.4.0
numpy 1.16.5
catkin

Error Log :
In file included from /home/spiri/ros_opencv_dependables/src/vision_opencv/cv_bridge/src/module_opencv2.cpp:35:0:
/home/spiri/ros_opencv_dependables/src/vision_opencv/cv_bridge/src/module.hpp: In function ‘int do_numpy_import()’:
/home/spiri/ros_opencv_dependables/src/vision_opencv/cv_bridge/src/module.hpp:39:5: warning: converting to non-pointer type ‘int’ from NULL [-Wconversion-null]
     import_array( );
     ^~~~~~~~~~~~
/home/spiri/ros_opencv_dependables/src/vision_opencv/cv_bridge/src/module_opencv2.cpp: At global scope:
/home/spiri/ros_opencv_dependables/src/vision_opencv/cv_bridge/src/module_opencv2.cpp:151:16: error: cannot declare variable ‘g_numpyAllocator’ to be of abstract type ‘NumpyAllocator’
 NumpyAllocator g_numpyAllocator;
                ^~~~~~~~~~~~~~~~
/home/spiri/ros_opencv_dependables/src/vision_opencv/cv_bridge/src/module_opencv2.cpp:98:7: note:   because the following virtual functions are pure within ‘NumpyAllocator’:
 class NumpyAllocator : public cv::MatAllocator
       ^~~~~~~~~~~~~~
In file included from /opt/include/opencv4/opencv2/core.hpp:59:0,
                 from /opt/include/opencv4/opencv2/core/core.hpp:48,
                 from /home/spiri/ros_opencv_dependables/src/vision_opencv/cv_bridge/include/cv_bridge/cv_bridge.h:43,
                 from /home/spiri/ros_opencv_dependables/src/vision_opencv/cv_bridge/src/module.hpp:22,
                 from /home/spiri/ros_opencv_dependables/src/vision_opencv/cv_bridge/src/module_opencv2.cpp:35:
/opt/include/opencv4/opencv2/core/mat.hpp:472:23: note: 	virtual cv::UMatData* cv::MatAllocator::allocate(int, const int*, int, void*, size_t*, cv::AccessFlag, cv::UMatUsageFlags) const
     virtual UMatData* allocate(int dims, const int* sizes, int type,
                       ^~~~~~~~
/opt/include/opencv4/opencv2/core/mat.hpp:474:18: note: 	virtual bool cv::MatAllocator::allocate(cv::UMatData*, cv::AccessFlag, cv::UMatUsageFlags) const
     virtual bool allocate(UMatData* data, AccessFlag accessflags, UMatUsageFlags usageFlags) const = 0;
                  ^~~~~~~~
/opt/include/opencv4/opencv2/core/mat.hpp:475:18: note: 	virtual void cv::MatAllocator::deallocate(cv::UMatData*) const
     virtual void deallocate(UMatData* data) const = 0;
                  ^~~~~~~~~~
/home/spiri/ros_opencv_dependables/src/vision_opencv/cv_bridge/src/module_opencv2.cpp: In function ‘int convert_to_CvMat2(const PyObject*, cv::Mat&)’:
/home/spiri/ros_opencv_dependables/src/vision_opencv/cv_bridge/src/module_opencv2.cpp:233:11: error: ‘class cv::Mat’ has no member named ‘refcount’
         m.refcount = refcountFromPyObject(o);
           ^~~~~~~~
/home/spiri/ros_opencv_dependables/src/vision_opencv/cv_bridge/src/module_opencv2.cpp: In function ‘PyObject* pyopencv_from(const cv::Mat&)’:
/home/spiri/ros_opencv_dependables/src/vision_opencv/cv_bridge/src/module_opencv2.cpp:254:12: error: ‘class cv::Mat’ has no member named ‘refcount’
     if(!p->refcount || p->allocator != &g_numpyAllocator)
            ^~~~~~~~
/home/spiri/ros_opencv_dependables/src/vision_opencv/cv_bridge/src/module_opencv2.cpp:261:36: error: ‘class cv::Mat’ has no member named ‘refcount’
     return pyObjectFromRefcount(p->refcount);

acxz, 2019-09-18:
See: Issue #272 and PRs #259 #288

ErictheSam, 2022-01-08:
Ran into the same problem, thanks for your solution!