ros-perception/vision_opencv #294: Dependency to GPL-licensed library JBIG-KIT

ralph-lange, 2019-10-02:
We analyzed the dependencies of the opencv3 package and discovered that it depends on libtiff, which again depends on libjbig, cf. https://github.com/ros-gbp/opencv3-release/blob/master/kinetic/package.xml.

While libtiff is licensed BSD-like, libjbig comes under GPL. The JBIG-KIT website at https://www.cl.cam.ac.uk/~mgk25/jbigkit/ states the intention that dynamic linking in a combined work makes this work to fall under GPL explicitly. At the same time, the website states an exception for the case that the JBIG-KIT features are not used, but this exception is not part of the actual license text.

A Google search revealed that others are aware of this issue, but did not provide a solution. 

Question 1: Can anybody confirm this dependency analysis?
Question 2: Would it be possible to provide an official OpenCV3 package without the libtiff functionalities?

[image]

mlauret, 2019-10-03:
I can confirm that libtiff is linked to jbig : 
```
roso@roso:~$ ldd /usr/lib/x86_64-linux-gnu/libtiff.so
	linux-vdso.so.1 =>  (0x00007ffcd31ad000)
	liblzma.so.5 => /lib/x86_64-linux-gnu/liblzma.so.5 (0x00007fa8b3c76000)
>>>>>>>>libjbig.so.0 => /usr/lib/x86_64-linux-gnu/libjbig.so.0 (0x00007fa8b3a68000)
	libjpeg.so.8 => /usr/lib/x86_64-linux-gnu/libjpeg.so.8 (0x00007fa8b380f000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007fa8b35f5000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007fa8b32ec000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fa8b2f22000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007fa8b2d1e000)
	/lib64/ld-linux-x86-64.so.2 (0x00007fa8b410d000)
```

You can also recompile libtiff without jbig support (https://gitlab.com/libtiff/libtiff) : 

```
roso@roso:~/libtiff/dist$ cmake -Djbig=off .
roso@roso:~/libtiff/dist$ make
roso@roso:~/libtiff/dist$ ldd libtiff/libtiff.so
	linux-vdso.so.1 =>  (0x00007ffefc1fd000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f61c63eb000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f61c61d1000)
	libjpeg.so.8 => /usr/lib/x86_64-linux-gnu/libjpeg.so.8 (0x00007f61c5f78000)
	liblzma.so.5 => /lib/x86_64-linux-gnu/liblzma.so.5 (0x00007f61c5d56000)
	libwebp.so.5 => /usr/lib/x86_64-linux-gnu/libwebp.so.5 (0x00007f61c5afa000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f61c5730000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f61c698e000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f61c552c000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f61c530f000)
```

ralph-lange, 2019-10-05:
@mlauret, thank you for confirming and for the hint on recompiling libtiff without jbig.

ijnek, 2022-09-12:
I recently became the maintainer of this repo, and I'm sorting through issues. @ralph-lange I'm going to guess it is, but could you confirm that this issue is still ongoing.

ralph-lange, 2022-09-13:
Hi @ijnek, thank you for taking over the maintenance of this repo. I can confirm that this issue is still present (just checked it in my Ubuntu 20.04 + ROS Noetic installation as well as my Ubuntu 22.04 + ROS 2 Humble installation):

- libopencv-imgcodecs4.2 depends on libtiff5
- libtiff5 depends on libjbig0
- The libjbig0 package meta information refers to https://www.cl.cam.ac.uk/~mgk25/jbigkit/, which explains that the JBIG-KIT is licensed under GPL as further explain above.