ros-perception/vision_opencv #303: Noetic and Python 3 Plan

sloretz, 2019-10-16:
Is there a plan for this repo to support Noetic and Python 3? In case you haven't seen it, there's [a tutorial on the ROS wiki about transitioning to Python 3](http://wiki.ros.org/UsingPython3).

It looks like `vision_opencv` [does not have a source entry](https://github.com/ros/rosdistro/blob/master/noetic/distribution.yaml) for Noetic [and is directly blocking 54 other repos](http://repositories.ros.org/status_page/blocked_source_entries_noetic.html?q=vision_opencv&s=6&r=1).  A source entry [signals to other packages it is being worked on](http://wiki.ros.org/UsingPython3/IdentifyDependencies#Determining_if_upstream_ROS_packages_support_Python_3), and makes it possible for downstream maintainers [to use `rosinstall_generator`](http://wiki.ros.org/UsingPython3/BuildUsingPython3#Download_source_code_of_dependencies) to download their dependencies while they work on Python 3 incompatibilities.

For starters, these rosdep keys will need to be switched to python 3 equivalents, and all the equivalents already exist:
* [`python` -> `python3`](https://github.com/ros/rosdistro/blob/c60825da29aec1cc7c21b3f48d3f1a4fe0c9e814/rosdep/python.yaml#L4905)
* [`python-opencv` -> `python3-opencv`](https://github.com/ros/rosdistro/blob/c60825da29aec1cc7c21b3f48d3f1a4fe0c9e814/rosdep/python.yaml#L5150)
* [`python-numpy` -> `python3-numpy`](https://github.com/ros/rosdistro/blob/c60825da29aec1cc7c21b3f48d3f1a4fe0c9e814/rosdep/python.yaml#L5143)

https://github.com/ros-perception/vision_opencv/blob/0bbebb73bea65eca856e732b919439853a04986b/cv_bridge/package.xml#L24-L25
https://github.com/ros-perception/vision_opencv/blob/0bbebb73bea65eca856e732b919439853a04986b/cv_bridge/package.xml#L31-L32
https://github.com/ros-perception/vision_opencv/blob/0bbebb73bea65eca856e732b919439853a04986b/cv_bridge/package.xml#L38

Rayman, 2019-10-16:
Something like this?
```
    <depend condition="$ROS_PYTHON_VERSION == '2'">python-numpy</depend>
    <depend condition="$ROS_PYTHON_VERSION == '3'">python3-numpy</depend>
```

sloretz, 2019-10-17:
Yup, minus the single quotes. The `package.xml` format will need to be bumped to `3` too.

See [rospy's package.xml](https://github.com/ros/ros_comm/blob/5da095d06bccbea708394b399215d8a066797266/clients/rospy/package.xml#L38-L43) and ros/genpy#104 for examples of conditional dependencies on `ROS_PYTHON_VERSION`.

AutonomousHansen, 2019-11-27:
Just commenting here to show my interest in also getting this to Noetic/Python3 compatibility

seanyen, 2020-02-25:
Commenting here for another Python3 related issue:

https://github.com/ros-perception/vision_opencv/blob/0bbebb73bea65eca856e732b919439853a04986b/cv_bridge/CMakeLists.txt#L11

On `Windows`, the naming scheme to look for `boost python 3` is little different from `Debian/Ubuntu`. In this particular case, I have to change it to look for `python37` to match my run-time. The similar issue is also discussed here: https://gitlab.kitware.com/cmake/cmake/issues/16391

We probably need a more robust way to find `boost python 3` module.

sloretz, 2020-03-12:
Hey @vrabaud, friendly ping. What's the plan for `vision_opencv` supporting Python 3 and ROS Noetic?

mjcarroll, 2020-03-28:
@ahcorde is going to work this, PR: https://github.com/ros-perception/vision_opencv/pull/323

sloretz, 2020-04-13:
Thanks for releasing! https://github.com/ros/rosdistro/pull/24373

closed 2020-04-13 by sloretz